bblanchon/ArduinoJson #403: Question - Not able to parse JSON

deejayspinz, 2016-12-23:
Hi,

Not sure where to post this question, so hopefully it's ok here.  I have successfully used your JSON parser on my weather project (thanks BTW!). I am trying to add another data stream to parse waste collection data.  I am using this [JSON](http://www.plastibots.com/garbageCalendarFormatted.json) which shows as valid JSON (run through an online validator).  The function is below (which works fine for both my other Weather Underground JSON parsing routines).

The issue is that `if (aLine.length() <= 1)` never evaluates and skip_headers does not get set to false. I also tried to set skip_headers to false at the start and dump the output.  The response dumps the bottom of the response (approx 70%) - as if it has ignored the top.      My buffer is: `respBuf[20*1024]; `. What am I doing wrong?  I also tried thinning down the JSON data and still the same issue.

```
void connectAndGetHaltonWaste()
{
  delay(2000);
  yield();

  myNextion.setComponentText("tStatusText", "Getting Waste calendar...");
  if (!client.connect(WasteHost, httpPort)) {
    if (debug) {Serial.println("Crap! cant connect to: Waste Calendar");}
    return;
  }  
  if (debug) {Serial.print("Connecting to: ");}  
  if (debug) {Serial.println(HR_WasteCollection);}  
    
  client.print(HR_WasteCollection);
  client.flush();  
  yield();
  if (debug) {Serial.print("Client connected WU - Server Request complete - ");}
  if (debug) {Serial.println(HR_WasteCollection);}

  int respLen = 0;
  bool skip_headers = true;
  while (client.connected() || client.available()) 
  {
    //if (debug) {Serial.println(F("Client is connnected... "));}
    if (skip_headers) 
    {
      String aLine = client.readStringUntil('\n');
      // Blank line denotes end of headers
      if (aLine.length() <= 1) 
      {
        skip_headers = false;
      }
      //if (debug) {Serial.println(F("Skip headers is true "));}
    }
    else 
    {
      int bytesIn;
      bytesIn = client.read((uint8_t *)&respBuf[respLen], sizeof(respBuf) - respLen);
      //if (debug) {Serial.print(F("bytesIn ")); Serial.println(bytesIn);}  //note, this will show debug info, but will cause ESP to crash.  USe for testing then comment out.
      if (bytesIn > 0) 
      {
        respLen += bytesIn;
        if (respLen > sizeof(respBuf)) respLen = sizeof(respBuf);
      }
      else if (bytesIn < 0) 
      {
        if (debug) {Serial.print(F("read error "));}
        if (debug) {Serial.println(bytesIn);}
      }
    }
    delay(1);
  }
  client.stop();
  yield();

  if (debug) {Serial.println("XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX");}  
  if (debug) {Serial.print("Getting Waste calendar n filling RespBuffer. Free Heap=  ");}  
  if (debug) {Serial.println(ESP.getFreeHeap());}  
  if (debug) {Serial.println("XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX");}  

    
  if (respLen >= sizeof(respBuf)) 
  {
    if (debug) {Serial.print(F("respBuf overflow "));}
    if (debug) {Serial.println(respLen);}
    delay(2000);
    return;
  }
  // Terminate the C string
  respBuf[respLen++] = '\0';
  if (debug) {Serial.print(F("respLen "));}
  if (debug) {Serial.println(respLen);}
  if (debug) {Serial.print(F("Raw Response content>>  "));}
  if (debug) { Serial.println(respBuf);}  //prints dump of all data - this also causes ESP to crash, so remove when testing verifies data
} 
```

bblanchon, 2016-12-24:
Hi,

Please understand that, as this issue is not related to ArduinoJson, here is not the right place for it.

The problem description doesn't give any clue of what is going wrong.
In particular, I don't understand what you're doing with `if (aLine.length() <= 1)`.
You should have a look at [JsonHttpClient.ino](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonHttpClient/JsonHttpClient.ino) and see how your problem differs.

I'm sorry but I cannot help you with this issue.

closed 2016-12-24 by bblanchon

deejayspinz, 2016-12-24:
Thanks for your comments.  Great library BTW. Thanks for putting it together.  

Merry Christmas.

 

Closed #403 <https://github.com/bblanchon/ArduinoJson/issues/403> .