bblanchon/ArduinoJson #404: Adafruit Wiced Feathe error: cannot convert 'err_t' to 'err_t (*)()

Andrewiski, 2016-12-24:
Wierd issue. Using Arduino IDE 1.8.0, I can not compile any code that includes ArduinoJson.h if the target board is an Adafruit Wiced Feather and the adafruit_feather.h is included.
Seems to be an issue with err_t. 

Not sure if its with Adafruit library or yours.

Remove #include <ArduinoJson.h>  and it compiles.

Source is Here https://github.com/adafruit/Adafruit_WICED_Arduino
Product Info is here https://www.adafruit.com/product/3056

error is as follows

`In file included from C:\Users\adevries\AppData\Local\Arduino15\packages\adafruit\hardware\wiced\0.6.2\cores\maple/adafruit_feather.h:49:0,

                 from C:\Users\adevries\Documents\Arduino\Adafruit Wicked\HTTP\SocketIOClient\SocketIOClient.ino:21:

C:\Users\adevries\AppData\Local\Arduino15\packages\adafruit\hardware\wiced\0.6.2\cores\maple/adafruit_sdep.h: In member function 'err_t (* AdafruitSDEP::__errno())()':

C:\Users\adevries\AppData\Local\Arduino15\packages\adafruit\hardware\wiced\0.6.2\cores\maple/adafruit_sdep.h:67:45: error: cannot convert 'err_t' to 'err_t (*)()' in return

   err_t        errno       (void ) { return _errno; }

                                             ^

In file included from c:\users\adevries\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\errno.h:9:0,

                 from C:\Users\adevries\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson/JsonVariantImpl.hpp:16,

                 from C:\Users\adevries\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson.hpp:20,

                 from C:\Users\adevries\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson.h:8,

                 from C:\Users\adevries\Documents\Arduino\libraries\ArduinoJson/ArduinoJson.h:8,

                 from C:\Users\adevries\Documents\Arduino\Adafruit Wicked\HTTP\SocketIOClient\SocketIOClient.ino:20:`

Andrewiski, 2016-12-24:
I was able to fix it in the adafruit_sdep.h file and reported it as a bug to adafruit.

https://github.com/adafruit/Adafruit_WICED_Arduino/issues/66

bblanchon, 2016-12-24:
[Added to Compatibility issues](https://github.com/bblanchon/ArduinoJson/wiki/Compatibility-issues#adafruit-wiced)

Thanks for reporting.

closed 2016-12-24 by bblanchon