bblanchon/ArduinoJson #406: contain nestedArrays into a new Array

Pfannex, 2016-12-26:
Hi,

i'd like to contain some nestedArrays into a new Array.
Like this...

```
//fill SUB-ARRAY with nested Arrays
JsonArray& MyNestedArrays[5];
  MyNestedArrays[0] = root[0].asArray();
  MyNestedArrays[1] = root[1][2].asArray();
  .....
```
Is there a way to to this?

Happay xmas
Pf@nne

bblanchon, 2016-12-27:
Hi,

The simplest is to use another `JsonArray`:

```c++
JsonArray& MyNestedArrays = jsonArray.createArray();
MyNestedArrays.add(root[0]);
MyNestedArrays.add(root[1][2]);
```

But in term of speed, the best is to use pointers:

```c++
JsonArray* MyNestedArrays[5];
MyNestedArrays[0] = &root[0].asArray();
MyNestedArrays[1] = &root[1][2].asArray();
```

As usual, make sure that the `JsonBuffer` stays in memory as long as you need `MyNestedArrays`.

Regards,
Benoit

Pfannex, 2016-12-27:
Hi Benoit,

thanks for your Help!

The pointer-way seems to be the most efficient.
```
JsonArray* MyNestedArrays[5];
MyNestedArrays[0] = &root[0].asArray();
MyNestedArrays[1] = &root[1][2].asArray();

```
But how to get access to the items?

```
Serial.println(MyNestedArrays[0][0].asString()) ;
```

Regards,
Marco

bblanchon, 2016-12-27:
Serial.println((*MyNestedArrays[0])[0].asString()) ;

or

    Serial.println(MyNestedArrays[0]->get<char*>(0)) ;

Pfannex, 2016-12-27:
`Serial.println((*MyNestedArrays[0])[0].asString()) ;`

Even if I don't quite understand the syntax.
But it works great.....! Thank you!

Maybe you got a helpfuly hint to copy one Array into another... :-)
```
MyNestedArrays[1] = MyNestedArrays[0][0].asArray();
MyNestedArrays[1] = (*MyNestedArrays[0])[0].asArray();
```

bblanchon, 2016-12-28:
Explanation:

1. `MyNestedArrays` is an array of pointer to `JsonArray`
2. `MyNestedArrays[0]` is a pointer to `JsonArray`
3. `*MyNestedArrays[0]` is a `JsonArray`
4. `(*MyNestedArrays[0])[0]` is the first element of the `JsonArray`
5. `(*MyNestedArrays[0])[0].asString()` is the first element converted to a string

That is indeed a very complex expression, and I'm curious why you need to write that kind of things...

There is no easy way to copy a `JsonArray` to another `JsonArray`; you'll have to use iterators.

However, there is a helper to extract a C array from a `JsonArray`:

```c++
int arrayOfInteger[32];
myJsonArray.copyTo(arrayOfInteger);
```

bblanchon, 2016-12-29:
@Pfannex, I believe you're focusing too much on the solution and not enough about the problem.

If I understand correctly, you're just looking for a way to transform a string like `"Devicename/output/OUT01/state"` into an identifier?

Why using a string as the identifier? Could an integer do the job?

Why not using a flat JSON configuration like the following?

```json
{
  "Devicename/output/OUT01/state": "1.2.1.1"
}
```

Pfannex, 2016-12-29:
>  I believe you're focusing too much on the solution and not enough about the problem.

maybe you're right.....
I'm looking for a easy and efficient way to control the topics.
For one thing the topic structure should be easy to read by user, for another thing it should be easy to handle inside the code (iteratable and compareable).

I also was thinking about the usage of objects.....
But for me it looks inconsistent in relation to the tree structure.

```c
{
"p1":"Topic",
"p2":
	{
	"p2.1":
		{
		"p2.1.1":"Topic",
		"p2.1.2":
			{
			"p2.1.2.1":"Topic",
			"p2.1.2.2":"Topic"
			}
		}
	},
"p3":"Topic",
"p4":"Topic",
"p5":
	{
	"p5.1":
```

How ever, I will have a look for some other ways.

```c
int arrayOfInteger[32];
myJsonArray.copyTo(arrayOfInteger);
```
Is there also a way back form the arrayOfInteger to the jsonArray?

bblanchon, 2016-12-29:
Yes, it's `copyFrom()`

Pfannex, 2016-12-29:
```c
  int arrayOfInteger[32];
  JsonArray& A_01 = pub_root[2].asArray();
  Serial.println(A_01[0].asString());
  A_01.copyTo(arrayOfInteger);

  JsonArray& A_02 = pub_JsonBuffer.createArray(); 
  A_02.copyFrom(arrayOfInteger);
  
  Serial.println(A_02[0].asString()); // <--empty
```

bblanchon, 2017-01-03:
`copyTo()` and `copyFrom()` helps you deals with flat JSON arrays like `[1,2,3]`.

If you want to print the resulting JSON, you need to call `printTo()`:

     A_02[0].printTo(Serial);

Pfannex, 2017-01-04:
```
  String jsonFileBuffer = "[\"1\",\"2\",\"3\",[\"3.1\",\"3.2\",\"3.3\"]]";
  DynamicJsonBuffer JsonBuffer;
  JsonArray& root = JsonBuffer.parseArray(jsonFileBuffer);

  int arrayOfInteger[32];
  JsonArray& A_01 = root[3].asArray();  //flat Array ["3.1","3.2","3.3"]
  Serial.println(A_01[0].asString());  //prints "3.1"
  A_01.copyTo(arrayOfInteger);

  JsonArray& A_02 = pub_JsonBuffer.createArray(); 
  A_02.copyFrom(arrayOfInteger);
  
  Serial.println(A_02[0].asString()); // <--empty
  A_02[0].printTo(Serial); // <--empty
```
Here A_02 seems to be empty, is this the right way to create the new array?

```
  JsonArray& A_02 = JsonBuffer.createArray(); 
  A_02.copyFrom(arrayOfInteger);

```

bblanchon, 2017-01-05:
You're welcome @Pfannex, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2017-01-05 by bblanchon