bblanchon/ArduinoJson #408: Need better way to convert JsonObject to uint8_t array

heman4t, 2016-12-29:
Hi, I am curious if is there any better way to convert `JsonObject` to `uint8_t array`?

My current function to send final `Response (i.e. JsonObject)` to hub via XBee radio.

```
uint8_t response[512];

void toOutside(ArduinoJson::JsonObject &Response) {
    String outgoing;
    Response.printTo(outgoing);
    outgoing.toCharArray((char *) response, outgoing.length() + 1);
    XBeeAddress64 addr64 = XBeeAddress64(0x0, 0x0);
    ZBTxRequest zbTx = ZBTxRequest(addr64, response, outgoing + 1);
    xbee.send(zbTx);
#if defined(debug)
    debug.println(F("[OUTPUT]"));
    debug.println((char *)response);
    debug.println();
#endif
}
```

ZBTxRequest is `ZBTxRequest(const XBeeAddress64 &addr64, uint8_t *payload, uint8_t payloadLength);`

heman4t, 2016-12-29:
I think its, `Response.printTo((char *) response, Response.measureLength());`

closed 2016-12-29 by heman4t

bblanchon, 2016-12-29:
The right answer is:

```c++
char response[512];
Response.printTo(response);
```

If, for some reason, you need a `uint8_t` array, you'll have to cast the pointer:

```c++
uint8_t response[512];
Response.printTo((char*)response, sizeof(response));
```