bblanchon/ArduinoJson #410: Array inside a JSON

pydesai, 2016-12-29:


bblanchon, 2016-12-30:
Just call `createNestedArray()` as in the [example](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonGeneratorExample/JsonGeneratorExample.ino#L46).
Then call `createNestedObject()` on the new array.

See:

* [ArduinoJson: Encoding JSON](https://github.com/bblanchon/ArduinoJson/wiki/Encoding%20JSON)
* [API Reference: JsonObject::createNestedArray](https://github.com/bblanchon/ArduinoJson/wiki/API%20Reference#jsonobjectcreatenestedarray)

@pydesai, I believed this was a well-covered topic in the documentation; I'm quite surprised you didn't find it. Do you have any suggestion to improve the documentation?

pydesai, 2016-12-30:
@bblanchon : Thanks for the prompt reply. I just skimmed through the examples topic in Wiki and found one with array but none with nested array containing json elements so posted here.

closed 2016-12-30 by pydesai