bblanchon/ArduinoJson #411: Error reading json string with percent "%" character

simpaltek, 2017-01-05:
does anyone know why code would error out when trying to parse json string that has a percent "%" character in it?  The percent character "%" is in the value not the field.

example

{
   "field": "this% will error out"
}

thanks,

scott

bblanchon, 2017-01-05:
Hi @simpaltek,

[This test](https://github.com/bblanchon/ArduinoJson/blob/5fbc36cc6a04d73ea6bf108532aa8610ff2f643b/test/Issue411.cpp#L12-L16) passes on all platforms:

```c++
char json[] = "{\"field\":\"this% will error out\"}";
DynamicJsonBuffer jsonBuffer;
JsonObject& root = jsonBuffer.parseObject(json);
ASSERT_TRUE(root.success());
ASSERT_STREQ("this% will error out", root["field"]);
```

Can you post a sample program ?

bblanchon, 2017-01-10:
Hi Scott,
Did you make any progress?

simpaltek, 2017-01-10:
Test it again and it did not work.  I am using version v5.6.2

my code:

  StaticJsonBuffer<MAX_JSON_CONFIG_ARDUINOJSON_BUFFER_SIZE> parseJsonBuffer;
  char* bodyCharArray = strdup(this->_http.arg("plain").c_str());
  JsonObject& parsedJson = parseJsonBuffer.parseObject(bodyCharArray); 
  if (!parsedJson.success()) {
    this->_interface->logger->logln(F("✖ Invalid or too big JSON"));
    String errorJson = String(FPSTR(PROGMEM_CONFIG_JSON_FAILURE_BEGINNING));
    errorJson.concat(F("Invalid or too big JSON\"}"));
    this->_http.send(400, FPSTR(PROGMEM_CONFIG_APPLICATION_JSON), errorJson);
    return;
  }

it works if I send without "%".

Any ideas i can try?

scott

bblanchon, 2017-01-10:
Hi Scott,

I believe the issue is a memory corruption, which could be due to:

* a stack overflow due to a too big `StaticJsonBuffer`
* a heap exhaustion due to the memory leak caused by `strdup()`

I suggest that you upgrade to ArduinoJson 5.8.0 and change your program for something like:

```c++
DynamicJsonBuffer jsonBuffer(MAX_JSON_CONFIG_ARDUINOJSON_BUFFER_SIZE);
JsonObject& parsedJson = jsonBuffer.parseObject(_http.arg("plain"));
if (!parsedJson.success()) {
  // ...
}
```

You can use the [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/) to find the value of `MAX_JSON_CONFIG_ARDUINOJSON_BUFFER_SIZE`.
In this case, you need to add the "Additional bytes for input duplication" since the content of `_http.arg("plain")` is duplicated.

tigergroup, 2017-01-10:
I will try that, but why would it work without the "%" character?  That part confused me.

scott

bblanchon, 2017-01-10:
That's the thing with memory corruption; you never know what's going to happen.

Maybe I'm wrong, but you need to try that before moving on.

tigergroup, 2017-01-11:
I could not try 5.8.0 due to some compile issues, but I did try 5.7.3 and same result.  Also, I tested with other characters such as "&", same issue.

used this code snippet:

DynamicJsonBuffer jsonBuffer(MAX_JSON_CONFIG_ARDUINOJSON_BUFFER_SIZE);
JsonObject& parsedJson = jsonBuffer.parseObject(_http.arg("plain"));
if (!parsedJson.success()) {
  // ...
}

Still received parsing error.

btw, I am compiling using Arduino 1.6.9 IDE for esp8266 chip.

any other suggestions?

scott

bblanchon, 2017-01-11:
I'm sorry about that; what was the compilation error?

Can you create a [gist](https://gist.github.com/) with a complete program? I have an ESP8266 with me to test it.

tigergroup, 2017-01-11:
ok, I figured out the problem.  It is not ArdruinoJson issue.  I am sending a json string via a http post.  Because it has a % in it, esp8266 web library thinks it is encoded which it is not.  So my resolution is to just do a url encoding and it works fine.

I apologize for the rabbit hunt I have caused, and I do appreciate your efforts in trying to resolve my problem.

Sincerely,

scott

bblanchon, 2017-01-12:
You're welcome Scott, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2017-01-12 by bblanchon

simpaltek, 2017-01-12:
Thanks Benoit, I starred the project.  You did a great job in writing it.

scott