bblanchon/ArduinoJson #413: Issue with JSON element not correctly formatted.

deejayspinz, 2017-01-06:
I have a project pulling data from Weather Underground. It's been working flawlessly for weeks.  However today, WU published JSON data that is messing it up. Wondering if the library handles non-quoted strings or if I have to do something in my code.  The offending item is dewpoint_c which is passing :null without quotes.

`"dewpoint_string":"NA",
"**dewpoint_c":null,**`

My codes results in an ESP WDT reset (yields are in there already):
`     strcpy(dewpointCurr, current["dewpoint_c"]);  <<fails here with WDT reset.
      if (debug) {Serial.print(" Current Dewpoint: ");Serial.print(dewpointCurr); Serial.println(" C (dewpoint), ");}`

bblanchon, 2017-01-06:
`null` is a valid JSON value and ArduinoJson translate it into a `NULL`.
This is why you get an access violation when calling `strcpy()`.

deejayspinz, 2017-01-06:
Ah.. I see. Thanks.  For anyone else who runs into this, fixed as follows:

strcpy(dewpointCurr, checkNulls(current["dewpoint_c"]));

const char *checkNulls(char const* val)
{
  const char *pt = val;
  
  if (strlen(pt) == 0)
  {
    return "";
  }
  else
  {
    return val;
  }
}

bblanchon, 2017-01-06:
```c++
const char* checkNulls(const char* s) {
    return s ? s : "";
}
```

but I think the Arduino's `String` class deals correcly with nulls, so this should work:

```c++
String dewpointCurr = checkNulls(current["dewpoint_c"];
```

deejayspinz, 2017-01-06:
Thanks for the tip.  Can you help me understand this statement? I assume its a comparison but not sure how it reads?  

`return s ? s : "";`

bblanchon, 2017-01-06:
It's a combination of the [conditional operator](https://en.wikipedia.org/wiki/%3F:) and implicit conversion from pointer to `bool`.
It's equivalent to:

```c++
if (s != NULL) 
  return s;
else
  return "";
```

deejayspinz, 2017-01-06:
Cool. Thx for your help. Here's what [I have done](https://goo.gl/Pt8Wxb) using your JSON parser (thx!)

bblanchon, 2017-01-06:
Congratulations 👍 
Please add it the [Wiki: Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

closed 2017-01-06 by bblanchon