bblanchon/ArduinoJson #414: 'class ArduinoJson::JsonBuffer' has no member named 'parseObject' in 5.8.0

rusboard, 2017-01-09:
Hello, can't say exactly how to say it, but after update to 5.8.0 my code not working. Can't do parseObject if this object was read from function param.

This example working in 5.7.3 and not working in 5.8.0. Please help understand what was changed and how to fix it?

```
#include <ArduinoJson.h>

JsonObject& getSubscribedUsers(JsonBuffer& jsonBuffer) {
  String example = "{\"Name\":\"John\"}";

  JsonObject& users = jsonBuffer.parseObject(example);

  return users;
}

void setup() {
  Serial.begin(9600);

  DynamicJsonBuffer jsonBuffer;
  JsonObject& users = getSubscribedUsers(jsonBuffer);

  users.printTo(Serial);
}

void loop() {

}
```

bblanchon, 2017-01-09:
Hi  @rusboard,

Sorry. My bad.
I forgot to add the "breaking changes" section 🙄 
See https://github.com/bblanchon/ArduinoJson/releases/tag/v5.8.0

Please tell me if it's clear enough, so I can copy this text to the CHANGELOG where it would be easier to find.

Thank you very much for reporting this.

rusboard, 2017-01-09:
Clear enough, Thank you.

closed 2017-01-09 by rusboard

bblanchon, 2017-01-10:
You're welcome @rusboard, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink: