bblanchon/ArduinoJson #415: error: invalid conversion from 'volatile void*' to 'const void*'

akastulin, 2017-01-10:
Hi Guys,
My board is ESP8266.
After upgrading to 5.8 start getting the following error:
In file included from /Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/ArduinoStream.hpp:11:0,
                 from /Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/StringTraits.hpp:37,
                 from /Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/DynamicStringBuilder.hpp:11,
                 from /Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/JsonPrintable.hpp:13,
                 from /Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../JsonVariantBase.hpp:12,
                 from /Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../JsonVariant.hpp:16,
                 from /Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../JsonBuffer.hpp:14,
                 from /Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/JsonParser.hpp:10,
                 from /Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonBufferBase.hpp:10,
                 from /Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 from /Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson.hpp:10,
                 from /Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson.h:8,
                 from /Users/user/Documents/Arduino/libraries/ArduinoJson/ArduinoJson.h:8,
                 from sketch/Client.h:4,
                 from /Users/user/Documents/Arduino/NodeMCUSensorsV4.6/NodeMCUSensorsV4.6.ino:11:
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/../TypeTraits/IsBaseOf.hpp: In instantiation of 'class ArduinoJson::TypeTraits::IsBaseOf<Stream, const volatile int>':
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp:165:56:   required from 'bool ArduinoJson::JsonObject::setNodeAt(TStringRef, TValueRef) [with TStringRef = const char*; TValueRef = const volatile int&]'
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp:78:19:   required from 'bool ArduinoJson::JsonObject::set(const TString&, const TValue&) [with TValue = volatile int; TString = const char*]'
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObjectSubscript.hpp:42:5:   required from 'ArduinoJson::JsonObjectSubscript<TKey>& ArduinoJson::JsonObjectSubscript<TKey>::operator=(const T&) [with T = volatile int; TString = char [12]]'
/Users/user/Documents/Arduino/NodeMCUSensorsV4.6/NodeMCUSensorsV4.6.ino:424:21:   required from here
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/../TypeTraits/IsBaseOf.hpp:26:57: error: invalid conversion from 'const volatile void*' to 'const void*' [-fpermissive]
     value = sizeof(probe(reinterpret_cast<TDerived *>(0))) == sizeof(Yes)
                                                         ^
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/../TypeTraits/IsBaseOf.hpp:22:14: error:   initializing argument 1 of 'static char (& ArduinoJson::TypeTraits::IsBaseOf<TBase, TDerived>::probe(const void*))[2] [with TBase = Stream; TDerived = const volatile int; ArduinoJson::TypeTraits::IsBaseOf<TBase, TDerived>::No = char [2]]' [-fpermissive]
   static No &probe(const void *);
              ^
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/../TypeTraits/IsBaseOf.hpp: In instantiation of 'class ArduinoJson::TypeTraits::IsBaseOf<Stream, volatile int>':
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/StringTraits.hpp:19:8:   required from 'struct ArduinoJson::Internals::StringFuncs<const volatile int, void>'
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/StringTraits.hpp:22:8:   required from 'struct ArduinoJson::Internals::StringFuncs<const volatile int&, void>'
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp:165:56:   required from 'bool ArduinoJson::JsonObject::setNodeAt(TStringRef, TValueRef) [with TStringRef = const char*; TValueRef = const volatile int&]'
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObject.hpp:78:19:   required from 'bool ArduinoJson::JsonObject::set(const TString&, const TValue&) [with TValue = volatile int; TString = const char*]'
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonObjectSubscript.hpp:42:5:   required from 'ArduinoJson::JsonObjectSubscript<TKey>& ArduinoJson::JsonObjectSubscript<TKey>::operator=(const T&) [with T = volatile int; TString = char [12]]'
/Users/user/Documents/Arduino/NodeMCUSensorsV4.6/NodeMCUSensorsV4.6.ino:424:21:   required from here
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/../TypeTraits/IsBaseOf.hpp:26:57: error: invalid conversion from 'volatile void*' to 'const void*' [-fpermissive]
     value = sizeof(probe(reinterpret_cast<TDerived *>(0))) == sizeof(Yes)
                                                         ^
/Users/user/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/Deserialization/../Serialization/../StringTraits/../TypeTraits/IsBaseOf.hpp:22:14: error:   initializing argument 1 of 'static char (& ArduinoJson::TypeTraits::IsBaseOf<TBase, TDerived>::probe(const void*))[2] [with TBase = Stream; TDerived = volatile int; ArduinoJson::TypeTraits::IsBaseOf<TBase, TDerived>::No = char [2]]' [-fpermissive]
   static No &probe(const void *);
              ^

However adding the line #define ARDUINOJSON_ENABLE_ARDUINO_STREAM 0
muted the issue.
Thank you

bblanchon, 2017-01-10:
Hi@akastulin,

Thank you very much for reporting this bug.

Can you try the branch [issue415](https://github.com/bblanchon/ArduinoJson/tree/issue415) or [apply the fix manually](https://github.com/bblanchon/ArduinoJson/commit/57ddd190dd9bf645dd7511991204b2cfae774b38)?

closed 2017-01-11 by bblanchon

akastulin, 2017-01-11:
I'll try