bblanchon/ArduinoJson #416: Dynamic char[] creation fails on parseObject with "template argument deduction/substitution failed" error

rswift, 2017-01-10:
Hello,

So far, the library has been a joy to use. I just updated to 5.8.0 and started seeing problems, my code reads:
```Arduino
    char probePacket[probePacketLength + 1];
    memset(probePacket, 0, probePacketLength + 1);
    probeUdp.read(probePacket, probePacketLength);
    DynamicJsonBuffer jsonBuffer;
    JsonObject& probeJson = jsonBuffer.parseObject(probePacket);
```

and the error I get is lengthy, I'll include it fully below by the salient but seems to be:

> /Users/robert/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonBufferBase.hpp:66:15: note:   template argument deduction/substitution failed:
/Users/robert/Development/Arduino/WiFiTemperatureBroadcast/HallEffectBroadcast/fnProbeStatus.ino:18:63: note:   variable-sized array type 'char [(((sizetype)((ssizetype)probePacketLength)) + 1)]' is not a valid template argument
     JsonObject& probeJson = jsonBuffer.parseObject(probePacket);
                                                               ^
exit status 1
no matching function for call to 'ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>::parseObject(char [(((sizetype)((ssizetype)probePacketLength)) + 1)])'

If I modify the first line of code to something like:
```Arduino
    char probePacket[100 + 1];
```
there are no compilation errors.

Thanks in advance for any guidance - Robert...

The full compiler error is here:

> /Users/robert/Development/Arduino/WiFiTemperatureBroadcast/HallEffectBroadcast/fnProbeStatus.ino: In function 'void updateProbeStatus()':
fnProbeStatus:18: error: no matching function for call to 'ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>::parseObject(char [(((sizetype)((ssizetype)probePacketLength)) + 1)])'
     JsonObject& probeJson = jsonBuffer.parseObject(probePacket);
                                                               ^
/Users/robert/Development/Arduino/WiFiTemperatureBroadcast/HallEffectBroadcast/fnProbeStatus.ino:18:63: note: candidates are:
In file included from /Users/robert/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/DynamicJsonBuffer.hpp:10:0,
                 from /Users/robert/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson.hpp:10,
                 from /Users/robert/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson.h:8,
                 from /Users/robert/Documents/Arduino/libraries/ArduinoJson/ArduinoJson.h:8,
                 from /Users/robert/Development/Arduino/WiFiTemperatureBroadcast/HallEffectBroadcast/HallEffectBroadcast.ino:48:
/Users/robert/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonBufferBase.hpp:60:15: note: template<class TString> ArduinoJson::JsonObject& ArduinoJson::JsonBufferBase<TDerived>::parseObject(const TString&, uint8_t) [with TString = TString; TDerived = ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>]
   JsonObject &parseObject(
               ^
/Users/robert/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonBufferBase.hpp:60:15: note:   template argument deduction/substitution failed:
/Users/robert/Development/Arduino/WiFiTemperatureBroadcast/HallEffectBroadcast/fnProbeStatus.ino:18:63: note:   variable-sized array type 'char [(((sizetype)((ssizetype)probePacketLength)) + 1)]' is not a valid template argument
     JsonObject& probeJson = jsonBuffer.parseObject(probePacket);
                                                               ^
In file included from /Users/robert/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/DynamicJsonBuffer.hpp:10:0,
                 from /Users/robert/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson.hpp:10,
                 from /Users/robert/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson.h:8,
                 from /Users/robert/Documents/Arduino/libraries/ArduinoJson/ArduinoJson.h:8,
                 from /Users/robert/Development/Arduino/WiFiTemperatureBroadcast/HallEffectBroadcast/HallEffectBroadcast.ino:48:
/Users/robert/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonBufferBase.hpp:66:15: note: template<class TString> ArduinoJson::JsonObject& ArduinoJson::JsonBufferBase<TDerived>::parseObject(TString&, uint8_t) [with TString = TString; TDerived = ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>]
   JsonObject &parseObject(
               ^
/Users/robert/Documents/Arduino/libraries/ArduinoJson/include/ArduinoJson/JsonBufferBase.hpp:66:15: note:   template argument deduction/substitution failed:
/Users/robert/Development/Arduino/WiFiTemperatureBroadcast/HallEffectBroadcast/fnProbeStatus.ino:18:63: note:   variable-sized array type 'char [(((sizetype)((ssizetype)probePacketLength)) + 1)]' is not a valid template argument
     JsonObject& probeJson = jsonBuffer.parseObject(probePacket);
                                                               ^
exit status 1
no matching function for call to 'ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>::parseObject(char [(((sizetype)((ssizetype)probePacketLength)) + 1)])'

bblanchon, 2017-01-11:
Hi Robert,

Thank you for reporting this issue.

I totally forgot to test VLA (Variable Length Arrays).
I'm going to need a few days to fix this.

Sorry for the inconvenience.

rswift, 2017-01-11:
oops ;)

Thank you for the reply. I'll keep an eye out and when any updates are ready, I'll gladly test them with my code - although I'm guess unit testing would be more robust than my single scenario.

Thanks again - Robert...

closed 2017-01-15 by bblanchon

rswift, 2017-01-16:
Thanks for addressing this quickly. Compiler error vanished with 5.8.1 :)

bblanchon, 2017-01-16:
You're welcome Robert.
Thank you for using ArduinoJson.