bblanchon/ArduinoJson #417: pubnub and arduino son

chrisw85, 2017-01-11:
Hi there

thanks for putting together a really neat library. I'm hoping for a little guidance on how to use your library with PubNub. I've been using aJson but I'm running into memory issues and subsequently the arduino hangs. I've tried reworking some of the pubnub example scripts but I run into issues with using the client (WiFiClient doesn't seem to play nice with the PubNub_Base_Client). Anyways, hopefully you can shed some light on the subject and point me in the right direction.

Thanks

Chris

closed 2017-01-11 by chrisw85

bblanchon, 2017-01-11:
Hi Chris,
I saw you close the issue.
Did you find a solution?

chrisw85, 2017-01-11:
Hi Benoit

You know how these things go. Right after posting the issue i got it working. The problem was on the pubnub publishing side. I was JSON encoding the message before sending which is unnecessary as pubnub encodes the messages for you. I've included my code. It works only if there is one message at a time, otherwise it is not able to parse. This is fine for the meantime but I'd like to figure a way around this. Any ideas?

void loop()
{
	
	WiFiClient *client;

	Serial.println("waiting for a message (subscribe)");
	PubSubClient *sclient = PubNub.subscribe(channel);
    if (!sclient) return; // error
    char buffer[64]; size_t buflen = 0;
    while (sclient->wait_for_data()) {
        buffer[buflen++] = sclient->read();
    }
    buffer[buflen] = 0;
    sclient->stop();
    Serial.println(buffer);

StaticJsonBuffer<200> jsonBuffer;

JsonArray& root = jsonBuffer.parseArray(buffer);

  if (!root.success()) {
    Serial.println("parse failed");

  }else{
    Serial.println("parse success");
  }
  const char* server1 = root[0]["pwm"]["8"];
  const char* server2 = root[0]["sender"]["name"];
  
  Serial.print("sender is: ");
  Serial.println(server2);
  Serial.print("pwm is: ");
  Serial.println(server1);
  
	delay(1000);
}