bblanchon/ArduinoJson #418: Create JSON pair with null value

moodgorning, 2017-01-13:
Hi, first of all thanks for this great library. 
I am trying to interact with a thingsShadow on AWS IOT and at some point in the control flow (http://docs.aws.amazon.com/iot/latest/developerguide/thing-shadow-data-flow.html) it requires to transmit a null value in the mqtt json object like this:

`{
    "state":{
        "reported":{
            "color":"green"
        },
        "desired":null}
    }
}`

Now I couldn't for the life of me find a way to make that happen with the library, neither did I find anything in the code or docs. I tried:

`state["desired"] = NULL ` which yields:  `{
    "state":{
        "reported":{
            "color":"green"
        },
        "desired":0}
    }
}`

and

`state["desired"] = "null"`  which yields: `{
    "state":{
        "reported":{
            "color":"green"
        },
        "desired":"null"}
    }
}`

as well as 

`state.createNestedObject("desired");` which yields: `{
    "state":{
        "reported":{
            "color":"green"
        },
        "desired":{}}
    }
}`

All are not the correct way to represent a null value. Any help with this would be much appreciated, I guess I'm just missing something.
Cheers

bblanchon, 2017-01-13:
Hi Max,

There are two ways to serialize a `null`:

* either set the value to `(char*)0`,
* or set the value to `RawJson("null")`

Please apologize as this was not documented.
I added this [question in the FAQ](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-insert-a-null).

moodgorning, 2017-01-13:
Yes that worked perfectly, thanks a lot for the quick reply.

closed 2017-01-13 by moodgorning

bblanchon, 2017-01-13:
You're welcome Max.
Thank you for using ArduinoJson.