bblanchon/ArduinoJson #419: Decoding JSON error

VTeselkin, 2017-01-15:
when i try Decoding current json ={"status":"post","message":"canal","data":{"canal":[3,3,3,3,3,3,3,3],"canal_timer":[1,1,1,1,1,1,0,0]}}
I get Null !!! But it`s valid json!

bblanchon, 2017-01-15:
Hello Vadim,

What did you try?
Did you read the documentation?
I'm pretty sure it's something very simple, like a too small `JsonBuffer`.

See:

* [Examples: JsonParserExample.ino](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonParserExample/JsonParserExample.ino)
* [Getting Started: Decoding JSON](https://github.com/bblanchon/ArduinoJson/wiki/Decoding%20JSON)
* [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)
* [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/)

VTeselkin, 2017-01-15:
I`m send {"status":"post","message":"canal","data":{"canal":[3,3,3,3,3,3,3,3],"canal_timer":[1,1,1,1,1,1,0,0]}}
And try parse:
`  StaticJsonBuffer<200> jsonBuffer;
 JsonObject& data = jsonBuffer.parseObject(inString);
         if (!data.success()) {
            Serial.println("parseObject() failed");
            return;
          }
         Serial.println("parseObject() success");
         Serial.println(data["data"]["canal_timer"][0].asString());
         return;`
And this return null:  Serial.println(data["data"]["canal_timer"][0].asString());

bblanchon, 2017-01-16:
`asString()` returns `NULL` when the variant is not a string, this is the correct behavior.

I think you want to write:

```c++
Serial.println(data["data"]["canal_timer"][0].as<int>());
```

BTW, the `asString()` function has been deprecated for a long time, you should use `as<char*>()` instead.

Can you please tell why you believed `asString()` was the way to go?

VTeselkin, 2017-01-17:
I still have a problem with parsing JSON. It's my example:
```
			Serial.println(inString);
			JsonObject& data = jsonBuffer.parseObject(inString);
			if (!data.success()) {
				Serial.println(F("parseObject() failed"));
				return;
			}
			Serial.println(F("parseObject() success"));
```
And I have next result.
```
{"status":"post","message":"time_NTP","data":{"epoch":1484692006}}
parseObject() success
{"status":"post","message":"canal","data":{"canal":[3,3,3,3,3,3,3,3],"canal_timer":[1,1,1,1,1,1,0,0]}}
parseObject() failed
```
Why if I try use inner array I'm couldn't parse my json?

bblanchon, 2017-01-18:
Hello Vadim,

It looks like the `JsonBuffer` is too small.

See:

* [Getting Started: Decoding JSON](https://github.com/bblanchon/ArduinoJson/wiki/Decoding%20JSON)
* [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)
* [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/)

bblanchon, 2017-01-24:
Hello Vadim,
Did you solve your problem?
Can I close this issue?

VTeselkin, 2017-01-24:
Yes, you can close this issue, i had problem due to small Json buffer. Thank you!!

bblanchon, 2017-01-24:
You're welcome Vadim, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2017-01-24 by bblanchon