bblanchon/ArduinoJson #421: Parsing error with comment

hallard, 2017-01-16:
Docs mention that comments are allowed in 2 formats
```
// comment
```
 or
```
/* Comments */
```
but as soon as I put comment I can't parse file (from ESP8266 SPIFFS), when I remove them, all is fine, I tried
```json
{
  "gateway": {
    "rgb_luminosity": 25,
// 1=GRBW, 2=RGBW, 3=GRB, 4=RGB
    "rgb_led_type": 4,    
    "oled": 1,
  }
}
```
```json
{
  "gateway": {
    "rgb_luminosity": 25,
/* 1=GRBW, 2=RGBW, 3=GRB, 4=RGB */
    "rgb_led_type": 4,    
    "oled": 1,
  }
}
```
```json
{
  "gateway": {
    "rgb_luminosity": 25,
    "rgb_led_type": 4,    // 1=GRBW, 2=RGBW, 3=GRB, 4=RGB
    "oled": 1,
  }
}
```
```json
{
  "gateway": {
    "rgb_luminosity": 25,
    "rgb_led_type": 4,    /* 1=GRBW, 2=RGBW, 3=GRB, 4=RGB */
    "oled": 1,
  }
}
```
none of my implementation is working, what I'm doing wrong ?

And here below my reading implementation on ESP8266 SPIFFS

```cpp
/* ======================================================================
Function: read_config_gateway
Purpose : read json gateway config file
Input   : -
Output  : true if config ok, false otherwise
Comments: -
====================================================================== */
bool read_config_gateway(void) 
{
  const char cfg_file[] = "/cfg_gateway.json";
  bool ret = false;

  size_t size = check_config_file(cfg_file);

  if ( size ) {

    File configFile = SPIFFS.open(cfg_file, "r");

    // Allocate a buffer to store contents of the file.
    std::unique_ptr<char[]> buf(new char[size]);

    // We don't use String here because ArduinoJson library requires the input
    // buffer to be mutable. If you don't use ArduinoJson, you may as well
    // use configFile.readString instead.
    if ( configFile.readBytes(buf.get(), size) == size ) {
      DynamicJsonBuffer jsonBuffer;
      JsonObject& json = jsonBuffer.parseObject(buf.get());

      if (json.success()) {
        const char* data ;

        uint16_t mode = 0;
        mode = json["gateway"]["rgb_luminosity"];
        if (mode) {
          Debugf("JSON set RGB Luminosity to %d%%\r\n", mode);
          rgb_luminosity = mode;
        }

        mode = json["gateway"]["rgb_led_type"];
        if (mode) {
          Debugf("JSON set RGB LED Type to %d\r\n", mode);
          rgb_led_type = (RgbLedType_e) mode;
        }

        // All is fine
        ret = true;
      } else {
        DebuglnF("Failed to parse file");
      }
    } else {
      DebuglnF("Failed to read file content");
    }

    if (configFile) {
      configFile.close();
    }

  }
  return ret;
}
```

Each time I got this output

```
SPIFFS Mount succesfull
FS File: /cfg_wifi.json, size: 111
FS File: /cfg_module.json, size: 808
FS File: /index.htm, size: 4072
FS File: /edit.htm, size: 17179
FS File: /cfg_gateway.json, size: 191
file /cfg_wifi.json looks good (111 bytes)
JSON override SSID '' to 'CH2I-HOTSPOT'
JSON override PSK '' to 'Wireless@Ch2i'
file /cfg_gateway.json looks good (191 bytes)
Failed to parse file
```

bblanchon, 2017-01-17:
Hello Charles,

The bug has been [fixed in the master branch](https://github.com/bblanchon/ArduinoJson/commit/17a17c8957cc54146226706031833cfcecebb9e5) and will be available in v5.8.2 in a few days.

Thank you very much for reporting this bug.

Regards,
Benoit

closed 2017-01-17 by bblanchon

hallard, 2017-01-18:
Benoit,

Nickel, just pulled latest, will try it tomorrow 

Charles