bblanchon/ArduinoJson #422: Decoding JSON received with Serial

marlonrettore, 2017-01-18:
Good morning mate,

I'm getting a JSOn via Serial from another arduino, how do I do the decoding of this JSON, I could not do it. Could you give me an example of a serial JSON reception of his decoding?

Thanks, I'm waiting

bblanchon, 2017-01-18:
Hello @marlonrettore,

It's a bug.
I'm calling `Stream.read()` which doesn't take the timeout into account.

Thanks for reporting this.

Regards,
Benoit

marlonrettore, 2017-01-19:
Good morning mate

Would it look like this?

char json[] = Stream.read();

StaticJsonBuffer<200> jsonBuffer;

JsonObject& root = jsonBuffer.parseObject(json);

if (!root.success())
{
  Serial.println("parseObject() failed");
  return;
}
const char* sensor    = root["sensor"];
long        time      = root["time"];
double      latitude  = root["data"][0];
double      longitude = root["data"][1];

bblanchon, 2017-01-19:
Hello @marlonrettore,

I'll be even simpler, as no `char[]` is required:

```c++
JsonObject& root = jsonBuffer.parseObject(Serial);
```

But you'll need a bigger `JsonBuffer` to hold the copy of the input.

It's already fixed in the [issue422 branch](https://github.com/bblanchon/ArduinoJson/tree/issue422) if you want to try.

closed 2017-01-22 by bblanchon