bblanchon/ArduinoJson #424: Problem acquiring data from https and a structured body

maximo80, 2017-01-19:
HI,
I found your library very useful, I'm looking for this a long time ago but when I tried to check it out I cannot make it works for me:

This is he http I want to get data:
https://my.tado.com/mobile/1.9/getCurrentState?username=xxx@xxx.com&password=xxxxx
this is the fixed data is answering:
{
"success": true,
"operation": "HOME",
"preheating": false,
"preheatingTargetOperation": null,
"autoOperation": "HOME",
"operationTrigger": "SYSTEM",
"insideTemp": 20.28,
"setPointTemp": 20,
"controlPhase": "STEADY",
"currentUserPrivacyEnabled": false,
"currentUserGeoStale": null,
"deviceUpdating": false,
"homeId": 17285,
"heatingOn": true,
"internetGatewayType": "GW02",
"internetGatewayConnected": true,
"heatingMuscleType": "RU01",
"heatingMuscleConnected": true,
"insideTemperatureSensorType": "RU01",
"insideTemperatureSensorConnected": true,
"settingsEnabled": true,
"fallbackOperation": false,
"helpUrl": "https://support.tado.com/hc/",
"pendingDeviceInstallation": false
}
I put it in the server and resource:
const char* server = "my.tado.com"; // server's address
const char* resource = "/mobile/1.9/getCurrentState?username=xxxx&password=xxxxxx";

I configurated a fixed root object buffer as 24
And also I have changed root name to (for example: "operation")
This is the result:
JSON parsing failed!

I've read you FAQ and documentation but I cannot go on, sorry If it is very easy answer but I am desperate.
Thanks.

bblanchon, 2017-01-19:
Hi @maximo80,

The first things to check are:

1. Is the input correctly received (BTW, you forgot to show that part of the code)
2. Is the JsonBuffer big enough

See:

* [FAQ: Why parsing fails](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)

maximo80, 2017-01-19:
Hi @bblanchon ,
I've only changed this:
1. server/resource
const char* server = "my.tado.com"; // server's address
const char* resource = "/mobile/1.9/getCurrentState?username=xxxx&password=xxxxxx";
2. BUFFER
bool readReponseContent(struct UserData* userData) {
  // Compute optimal size of the JSON buffer according to what we need to parse.
  // This is only required if you use StaticJsonBuffer.
  const size_t BUFFER_SIZE =
      JSON_OBJECT_SIZE(24);    // the root object has 8 elements
      //+ JSON_OBJECT_SIZE(5)  // the "address" object has 5 elements
      //+ JSON_OBJECT_SIZE(2)  // the "geo" object has 2 elements
      //+ JSON_OBJECT_SIZE(3)  // the "company" object has 3 elements
      + MAX_CONTENT_SIZE;    // additional space for strings

3.parsing strings
// Here were copy the strings we're interested in
  strcpy(userData->name, root["operation"]);
  //strcpy(userData->company, root["company"]["name"]);
  // It's not mandatory to make a copy, you could just use the pointers
  // Since, they are pointing inside the "content" buffer, so you need to make
  // sure it's still in memory when you read the string

  return true;
}

// Print the data extracted from the JSON
void printUserData(const struct UserData* userData) {
  Serial.print("Name = ");
  Serial.println(userData->name);
  //Serial.print("Company = ");
  //Serial.println(userData->company);

Finally It looks to connect to server but it gets this error: JSON parsing failed!

Answers:
1. I think it is received but I don't know how to check it out.
2. I think it is enough there are 24 root objects in the body message

Thanks.

bblanchon, 2017-01-19:
I'm very surprised that the EthernetClients support HTTPS.
Did you try HTTP?

You should dump the received content to the Serial port.
For instance:

```c++
String s = client.readString();
Serial.println(s);
```

maximo80, 2017-01-19:
Maybe you are right
This is the response
Serial ready
Ethernet ready
Connect to my.tado.com
Connected
GET /mobile/1.9/getCurrentState?username=xx@hotmail.com&password=xx
tml>
<head><title>301 Moved Permanently</title></head>
<body bgcolor="white">
<center><h1>301 Moved Permanently</h1></center>
<hr><center>nginx</center>
</body>
</html>

JSON parsing failed!
Disconnect
Wait 60 seconds

bblanchon, 2017-01-20:
Hello @maximo80,

You could print the HTTP headers in the same manner.
I bet it redirects to the HTTPS version.

Most Arduino's boards are unable to do HTTPS, but a few can.
For instance:

* the Arduino Yùn can do HTTPS on the Linux side,
* the ESP8266 has a WifiClientSecure to do that.

I'm closing this issue as it's not related to ArduinoJson.
Please reopen if I'm wrong or open a new issue if you need help on another problem.

Regards,
Benoit

closed 2017-01-20 by bblanchon