bblanchon/ArduinoJson #425: Redefinition of StringTraits<const char*, void>'

porkyneal, 2017-01-19:
Hi, I am using the esp32 Arduino platform in Arduino IDE 1.6.10.

If I try and compile any sketch, including a blank, with ArduinoJson I get these warnings:

```
In file included from software\arduino\libraries\arduino_658289\include\arduinojson\stringtraits\stringtraits.hpp:41:0,

                 from software\arduino\libraries\arduino_658289\include\arduinojson\Serialization/DynamicStringBuilder.hpp:11,

                 from software\arduino\libraries\arduino_658289\include\arduinojson\Serialization/JsonPrintable.hpp:13,

                 from software\arduino\libraries\arduino_658289\include\arduinojson\JsonVariantBase.hpp:12,

                 from software\arduino\libraries\arduino_658289\include\arduinojson\JsonVariant.hpp:16,

                 from software\arduino\libraries\arduino_658289\include\arduinojson\jsonbuffer.hpp:14,

                 from Software\Arduino\libraries\arduino_658289/include/ArduinoJson/Deserialization/JsonParser.hpp:10,

                 from Software\Arduino\libraries\arduino_658289/include/ArduinoJson/JsonBufferBase.hpp:10,

                 from Software\Arduino\libraries\arduino_658289/include/ArduinoJson/DynamicJsonBuffer.hpp:10,

                 from Software\Arduino\libraries\arduino_658289/include/ArduinoJson.hpp:10,

                 from Software\Arduino\libraries\arduino_658289/include/ArduinoJson.h:8,

                 from Software\Arduino\libraries\arduino_658289/ArduinoJson.h:8,

                 from AppData\Local\Temp\build26871afffb0c494bed0992070c328079.tmp\sketch\tm_ctrl.cpp:5:

software\arduino\libraries\arduino_658289\include\arduinojson\stringtraits\FlashString.hpp:13:8: error: redefinition of 'struct ArduinoJson::Internals::StringTraits<const char*, void>'

 struct StringTraits<const __FlashStringHelper*, void> {

        ^

In file included from software\arduino\libraries\arduino_658289\include\arduinojson\stringtraits\stringtraits.hpp:26:0,

                 from software\arduino\libraries\arduino_658289\include\arduinojson\Serialization/DynamicStringBuilder.hpp:11,

                 from software\arduino\libraries\arduino_658289\include\arduinojson\Serialization/JsonPrintable.hpp:13,

                 from software\arduino\libraries\arduino_658289\include\arduinojson\JsonVariantBase.hpp:12,

                 from software\arduino\libraries\arduino_658289\include\arduinojson\JsonVariant.hpp:16,

                 from software\arduino\libraries\arduino_658289\include\arduinojson\jsonbuffer.hpp:14,

                 from Software\Arduino\libraries\arduino_658289/include/ArduinoJson/Deserialization/JsonParser.hpp:10,

                 from Software\Arduino\libraries\arduino_658289/include/ArduinoJson/JsonBufferBase.hpp:10,

                 from Software\Arduino\libraries\arduino_658289/include/ArduinoJson/DynamicJsonBuffer.hpp:10,

                 from Software\Arduino\libraries\arduino_658289/include/ArduinoJson.hpp:10,

                 from Software\Arduino\libraries\arduino_658289/include/ArduinoJson.h:8,

                 from Software\Arduino\libraries\arduino_658289/ArduinoJson.h:8,

                 from AppData\Local\Temp\build26871afffb0c494bed0992070c328079.tmp\sketch\tm_ctrl.cpp:5:

software\arduino\libraries\arduino_658289\include\arduinojson\stringtraits\CharPointer.hpp:46:8: error: previous definition of 'struct ArduinoJson::Internals::StringTraits<const char*, void>'

 struct StringTraits<const char*, void> : CharPointerTraits {};

        ^

```

Any help would be hugely appreciated, love the library!

porkyneal, 2017-01-19:
Just seen the compatibility issues, oops 👎

closed 2017-01-19 by porkyneal

bblanchon, 2017-01-19:
[Link to the relevant section of the wiki](https://github.com/bblanchon/ArduinoJson/wiki/Compatibility%20issues#esp32)