bblanchon/ArduinoJson #426: format not supported

maximo80, 2017-01-20:
Hi again,

I could finally make a copy of https to http and now your script is reading the string.
I found It is not identifying well the header I suppose that the format is different.
Also It is not getting the structure I suppose as well that the format is different.

This is the http: http://afeelink.com/tado.php

I checked all buffers and there are right.

bblanchon, 2017-01-21:
Hi @maximo80,

I'll be glad to help you.
Unfortunately, there is not enough information in your message.
Can you please elaborate?

bblanchon, 2017-01-29:
@maximo80, any progress?
Should I close this issue?

maximo80, 2017-01-31:
Hi @bblanchon , I am trying to find other way but I'm having problems with parsing.
This is the reading:
{"success":true,"operation":"HOME","preheating":false,"preheatingTargetOperation":null,"autoOperation":"HOME","operationTrigger":"SYSTEM","insideTemp":19.91,"setPointTemp":20.0,"controlPhase":"HEATUP","currentUserPrivacyEnabled":false,"currentUserGeoStale":null,"deviceUpdating":false,"homeId":17285,"heatingOn":true,"internetGatewayType":"GW02","internetGatewayConnected":true,"heatingMuscleType":"RU01","heatingMuscleConnected":true,"insideTemperatureSensorType":"RU01","insideTemperatureSensorConnected":true,"settingsEnabled":true,"fallbackOperation":false,"helpUrl":"https://support.tado.com/hc/","pendingDeviceInstallation":false}

The difference with your http input is the newline your example have newline and this one doesn't have I think it is why it is not working properly.
What do you think is the modification I need to do to consider this?

thanks.

maximo80, 2017-02-05:
Hi, Close it.
I don’t know what it happens, but it doesn’t work well.

Thanks.

bblanchon, 2017-02-05:
@maximo80, I'd really like to help you, but you need to provide more details.
What doesn't work well?

closed 2017-02-11 by bblanchon