bblanchon/ArduinoJson #427: No control on when float numbers are written with exponent

hagai-shatz, 2017-01-21:
When serialising JSON with float/double values, there is control on how many digits to write after the decimal point, with default of 2.
However, if the number is bigger than 1000 or smaller than 0.001 it will always be serialized with exponent: 1234.5 to 1.23e7

I can track this hard coded behaviour to: https://github.com/bblanchon/ArduinoJson/blob/master/include/ArduinoJson/Serialization/JsonWriter.hpp#L95

This is not in line with Arduino Serial.print that does not add exponent.

bblanchon, 2017-01-21:
Hello Hagai,

What is the behavior that you expect?
Can you give concrete examples?

hagai-shatz, 2017-01-21:
I expect to have the same behaviour as Serial.print by default: write all the digits of the integer part. For example: the float number 123456.789 will be written as 123456.79 by default (since we have 2 digits after the decimal point by default).
It should be an option to provide the exponent threshold, not have it fixed to 3 (write with exponent if number is bigger than 10^3 or smaller than 10^-3). And the default threshold should be undefined to print as many digits as possible.

bblanchon, 2017-01-22:
I'm sorry Hagai, but I disagree.
I think that ArduinoJson does a much better job that Serial.print()
Here are a few examples:

| Input | Serial | ArduinoJson
| ------ | ------| ------|
| 3.14e10 | ovf | 3.14e10 |
| 3.14e-10 | 0.00 | 3.14e-10 |

Disabling the scientific notation by default is a bad idea.
Remember that a double goes up to `1.7976931348623157E+308`.
Without scientific notation, this gives `179769313486232000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.00`

Not only this is a huge amount of wasted memory, but it can also be a security issue if the developer hadn't anticipated such a long string.

Making the threshold configurable would increase the code size, which is only acceptable if the benefit is real.

Can you tell me why `123456.79` is better than `1.2345679e5`?

hagai-shatz, 2017-01-22:
Well... you are absolutely right.
My initial thought was to add precision parameter with a default value of 3 to keep the current behaviour (almost).
The code will look something like this, assuming another parameter `uint8_t precision = 3`:
```
    if (value > pow(10, precision) || value < pow(10, -precision) {
      powersOf10 = Polyfills::normalize(value);
      digits = precision - 1;
    } else {
      powersOf10 = 0;
    }
```
With default precision of 3 and digits set to 2 we get the same behaviour:

 Input | Output 
-------|--------
12.1234 | 12.12
234.2 | 234.2
1234.56 | 1.23e3
0.0001234 | 1.23e-4

When you want higher precision you can specify it, for example precision of 6:

 Input | Output | Exceptions
-------|--------|--------
1234.56 | 1234.56 |
123456789.123 | 1.23456e8 |
0.00012345678 | 1.23456e-4 |
12345.1234 | 12345.12 | Precision of 7 - digits is 2 by default 
12.1234 | 12.12 | Precision of 4 - digits is 2 by default

hagai-shatz, 2017-01-22:
> Can you tell me why 123456.79 is better than 1.2345679e5?

They both have the same precision so I don't really care, although the scientific representation consume additional 2 bytes: 'e5'.  
The current code will output the number 123456.79 as 1.23e5 unless I set digits to a higher value.
Maybe I will take this workaround for the time being and set digits to 12.

bblanchon, 2017-01-22:
That's true that those two additional bytes are annoying.
I'll see what I can do.

Meanwhile, you can use a `RawJson()` as a workaround.

markterrill, 2017-01-29:
Hi team, what do you do with arrays containing integers? ie [0,1,2,3]

Any integer causes a pass fail. My current guess being how parseNumberTo looks for a long or a float.
I need to be able to supply integers, how do I get that to work?

bblanchon, 2017-01-29:
@markterrill, I'm pretty sure this is not related to the current issue.
Please [open a new issue](https://github.com/bblanchon/ArduinoJson/issues/new), and please write a few lines of code to show the problem.
(I'll delete these two last messages once the new issue is created).

hagai-shatz, 2017-01-31:
Specifying 12 digits is a good workaround for now: `item.set("val", value, 12);`

I had difficulties using `RawJson()` with the `ESPAsyncWebServer` since I had no good way to keep the String/char* in memory once outside the handler, and by the time the server was serializing the Json, the String/char* was released.

andrei-ivanov, 2017-05-01:
Any chance to get this fixed? :-)

bblanchon, 2017-05-01:
@andrei-ivanov Yes and No.

Yes, I'm going to change the current behavior.
No, I'll not add a third parameter to `set(float,int)`.

Indeed, my plan is to remove the second parameter that specifies the number of decimal places.
This is an heritage from `Print` but it's not required anymore.
I want to remove this as I believe it's more a nuisance than a feature.

andrei-ivanov, 2017-05-01:
I just thought it would be nice to be able to have correct numbers serialized by default, no matter the notation and not having to use the `RawJson` workaround.

bblanchon, 2017-05-01:
My goals is to get the right number of decimals, without asking the user to specify it.
Unfortunately, it's a notoriously hard problem, and I need to keep the code small.

bblanchon, 2017-05-20:
Float serialization was completely rewritten in [v5.10.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.10.0)

closed 2017-05-20 by bblanchon

aj-ptw, 2017-06-17:
I have no ability to set the precision on a double that is an NTP time stamp in microseconds. 

I want `14977139448720123.0` but am getting `1e51`. How can i force this to be expanded. 

Likewise I'm using this for the new [OpenBCI Wifi Shield](www.github.com/openbci/openbci_wifi] (to put brainwaves on the internet) and the high precision 24bit ADS values converted into floats are on the microvolt level so numbers are `0.00001234567` and these are coming over like `0.00`.

devyte, 2017-06-17:
Out of curiosity, and without knowing anything about your application, why
are you using doubles and floats for that? I would use longs to keep
precision as long as possible, and convert to whatever units only at the
very end and only for display purposes to the user.

aj-ptw, 2017-06-20:
We want to be able to stream decimal values of a 24bit analog to digital converter. So we for example send a JSON object directly to the browser over web sockets. See [JSON out #13](https://github.com/openbci/openbci_wifi/issues/13)

devyte, 2017-06-20:
I guessed that, but why do you need to stream a float/double? Json is
inherently a text format, and float/doubles with a wide range of values,
like what you posted, don't play nice when printing to string. This is not
specific to the ESP, but a generic issue with how float/doubles work.

Try streaming the raw long value and convert to float in the browser, if
you really need floats.

aj-ptw, 2017-06-20:
so you think convert to nano volts (we are measuring micro volts) and then converting to longs?

aj-ptw, 2017-06-20:
i understand how floats/doubles work, using a text transfer protocol I accept the loss of precision, but how can I say got out 10 places?

bblanchon, 2017-06-25:
ArduinoJson 5.10 supports up to 9 digits after the decimal point.
For that you need to enable support for doubles with [ARDUINOJSON_USE_DOUBLE](https://bblanchon.github.io/ArduinoJson/api/config/use_double/)

aj-ptw, 2017-06-25:
Thank you! I am using longs where my voltages are in nano volts so I move the precision to the other side which should work just fine. I love ArduinoJson! Keep up the great work!

tsaitsai, 2017-10-08:
Just wanted to chime in to say that the higher precision support for ArduinoJson makes GPS coordinates much easier to work with.  Thanks for the update.

bblanchon, 2017-10-09:
Thanks, @tsaitsai! That kind of comment is scarce.
It's good to get positive feedback, especially since it's been a lot of work to implement this feature correctly.
By the way, I documented the process in this blog post:
https://blog.benoitblanchon.fr/lightweight-float-to-string/

AvatarRolon, 2018-03-12:
Excuse me, my doubt is that ArduinoJson is actually compatible with Ethernet Library, because i can not send data from arduino ethertnet to my server.
Explanation:
The object json is pretty well formed only when the variables for ethernet are not present, but when variables for ethernet are present, the object json is not formed as well as without variables, can you tell me why is passing this?
example with out variable for ethernet:
{
  "coordinador": "013A20040D95BE0",
  "dispositivo": "013A20040B7D712",
  "fecha": "12/12/12",
  "hora": "12:12:12,
  "sensores": [
    {
      "nombre": "DHT",
      "datos": {
        "temperatura": 10.5,
        "humedad": 10.5
      }
    },
    {
      "nombre": "Higometro1",
      "datos": {
        "humedad": 273
      }
    },
    {
      "nombre": "Higometro2",
      "datos": {
        "humedad": 254
      }
    },
    {
      "nombre": "Higometro3",
      "datos": {
        "humedad": 263
      }
    },
    {
      "nombre": "Higometro4",
      "datos": {
        "humedad": 281
      }
    }
  ]
}

With ethernet variables
{
  "coordinador": "013A20040D95BE0",
  "dispositivo": "013A20040B7D712",
  "fecha": "165/165/2165",
  "hora": "165:165:85",
  "sensores": [
    {
      "nombre": "DHT",
      "datos": {
        "temperatura": NaN,
        "humedad": NaN
      }
    },
    {
      "nombre": "Higometro1",
      "datos": {
        "humedad": 273
      }
    },
    {
      "nombre": "Higometro2",
      "datos": {
        "humedad": { }
    }
  ]
}

Sure i can give you my code, if it help, thanks