bblanchon/ArduinoJson #429: Hello,

apprafael, 2017-01-24:


apprafael, 2017-01-24:
Hello,

I'm trying to get a json from Serial but i can't figure out where is the problem.

This code works prety well:

`      char json[1024];
       strcpy(json,  "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}");

      StaticJsonBuffer<200> jsonBuffer;

      JsonObject& root = jsonBuffer.parseObject(json);

      const char* sensor = root["sensor"];

      Serial.println(sensor);`

but in this one, i can't take the json data:

`    if(Serial.available())
    {

       char json[1024];
       strcpy(json,  Serial.readString().c_str());

      StaticJsonBuffer<200> jsonBuffer;

      JsonObject& root = jsonBuffer.parseObject(json);

      const char* sensor = root["sensor"];

      Serial.println(sensor);

      
      
    }`

Thanks,

bblanchon, 2017-01-24:
Hello Raphael,

As stated in the [FAQ](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#case-1-the-input-is-not-a-valid-json), the number reason for failing parse is a bad input.
So the first thing to do is to `Serial.println(json)` before calling `parseObject()`.

If you upgrade to ArduinoJson 5.8.2, you can simplify to:

```c++
DynamicJsonBuffer jsonBuffer(1024);
JsonObject& root = jsonBuffer.parseObject(Serial);
const char* sensor = root["sensor"];
Serial.println(sensor);
```

Note that I made the `JsonBuffer` bigger so that it can hold a copy of the input.
I also switched to a `DynamicJsonBuffer` because I felt that 1024 may be a bit too much for the stack, but you can switch back to `StaticJsonBuffer` if that makes sense for your project.

bblanchon, 2017-01-25:
You're welcome, Rafael.
Did you figure out what the problem was?

anatom74, 2018-03-01:
> If you upgrade to ArduinoJson 5.8.2, you can simplify to:
DynamicJsonBuffer jsonBuffer(1024);
JsonObject& root = jsonBuffer.parseObject(Serial);
const char* sensor = root["sensor"];
Serial.println(sensor);

This one doesnt't work for me. When does parseObject need to be called ? 
@apprafael - would really see your orking code if possible

ocan23, 2018-04-11:
Hi bblanchon. In my case, I need to write and read a JSON data from serial. 

I have to do that in the same code. 

Right now I have the code for send json message to a raspberry pi by serial. This is my code : 

`#include <ArduinoJson.h>

void setup()
{ 
  Serial.begin (9600); //Abre el puerto serie y se fija la velocidad en baudios
}
void loop()
{
  int value_sensor=analogRead(A0); //Lee el pin de entrada analógico AO
  float voltage=value_sensor*(5.0/1023.0); 
  float oxygen=10.0/4.0*(voltage-1.0);

  DynamicJsonBuffer jBuffer;
  JsonObject& root= jBuffer.createObject();
  root ["Voltaje"]=(voltage);
  root ["Oxygen"]=(oxygen);
  root.prettyPrintTo(Serial);
  Serial.println();
  delay(10000);
  
  char json[]=Serial
}`

But now I have to read at the same time, so I have to do a read a message like that:

{
"Control": 0.590909
}

Can you help me with the code please? I´m not clear how I have to make the reading

bblanchon, 2018-04-12:
Hi @ocan23 

To parse a JSON document from the Serial, you need to pass `Serial` to `parseObject()`

```c++
JsonObject& root = jBuffer.parseObject(Serial);
```

Now, it's very important that you do the serialization and the deserialization in different scopes, preferably in different functions, so as to use a fresh `JsonBuffer` and not to reuse the existing one.

See also:

* [FAQ: Can I parse data from a stream?](https://arduinojson.org/faq/can-i-parse-data-from-a-stream/?utm_source=github&utm_medium=issues)
* [FAQ: How to reuse a JsonBuffer?](https://arduinojson.org/faq/how-to-reuse-a-jsonbuffer/?utm_source=github&utm_medium=issues)

Regards,
Benoit

ocan23, 2018-04-17:
Thank you for answered @bblanchon . I didn´t understand when you said ''it's very important that you do the serialization and the deserialization in different scopes, preferably in different functions, so as to use a fresh JsonBuffer and not to reuse the existing one.''

I created two jsonBuffer, jBuffer and jsonBuffer. It is wrong? This is my code 

#include <ArduinoJson.h>

void setup()
{ 
  Serial.begin (9600); 
}
void loop()
{
  int value_sensor=analogRead(A0); 
  float voltage=value_sensor*(5.0/1023.0); 
  float oxygen=10.0/4.0*(voltage-1.0);
  

  DynamicJsonBuffer jBuffer;
  JsonObject& root= jBuffer.createObject();
  root ["Voltaje"]=(voltage);
  root ["Oxygen"]=(oxygen);
  root.prettyPrintTo(Serial);
  Serial.println();

  
  if (Serial.available())
  {
    DynamicJsonBuffer jsonBuffer;
    jsonObject& root=jsonBuffer.parseObject(Serial)
    int control=root["Control"]
  }
  delay(10000);
    
}

ocan23, 2018-04-25:
Hi bblanchon, sorry for not answer before, I was very busy.

I modified my code after read [https://arduinojson.org/faq/whats-the-best-way-to-use-the-library/?utm_source=github&utm_medium=issues](url). 

Now is it: 
`
#include <ArduinoJson.h>
const int AnalogOutPin=11; //Definimos el pin 11 como pin analogico de salida PWM
int OutPutValue=51; // valor del PWM iniciado a un 20% para obtener los 4mA como minimo
int Control=0;
int IntervaloRead=10000;
long previousMillisRead = 0;

void setup() {
  Serial.begin(9600);
}

void loop()
{
  SendSensorData();
  ReadControl();
  OutPutValue=(204./100.0)*Control +51.0; //Si control es 100%, outputValue es 255, que son 5V y con el modulo tenemos una señal de 20mA
  analogWrite(AnalogOutPin,OutPutValue);
}

void SendSensorData()
{
  unsigned long currentMillis = millis();
   if(currentMillis - previousMillisRead > IntervaloRead) {
    previousMillisRead = currentMillis;   
    //Lectura del sensor. Se recibe 4-20mA y se pasa a 0-5V
    int value_sensor=analogRead(A0);    //Lee el pin de entrada analógico AO
    float voltage=value_sensor*(5.0/1023.0);    //Convertimos la lectura analógica (0-1023, por ser 10 bits) y lo paso a 0-5V
    float oxygen=10.0/4.0*(voltage-1.0);    //Escalamos el voltaje a nivel de oxigeno
    //float oxygen=0.5*voltage;
    //Creacion del mensaje JSON y se imprime por puerto serie
    DynamicJsonBuffer jBuffer;
    JsonObject& root= jBuffer.createObject();
    root ["Voltaje"]=(voltage);
    root ["Oxygen"]=(oxygen);
    root.prettyPrintTo(Serial);
    Serial.println();   
  }
}
int ReadControl()
{
  //Leer puerto serie en formato json y extraer el valor de la señal de control. Sera un porcentaje del uso de la bomba 0-100% siendo 0 un 4mA y un 100 un 20mA
    if (Serial.available())
    {
      DynamicJsonBuffer jsonBuffer;
      JsonObject& root=jsonBuffer.parseObject(Serial);
      const int Control=root["Control"]; //Obtengo un valor de 0 a 100 en porcentaje
      return Control;
    }
}`

I can´t read the signal by my Serial, but I think the code is well.

vrcjlb, 2018-05-04:
Hello @bblanchon,
I have the same problem, I do not know why I can not read my serialized data. The solution you indicate can be applied just as I read serialized data like this:
JsonArray& arr = jsonBuffer.parseArray(Serial1);

In advance thanks for the help

bblanchon, 2018-05-05:
What does `Serial.println(jsonSerial);` prints?
Is `jsonBuffer` global?
Is `jsonBuffer` big enough?