bblanchon/ArduinoJson #430: parseObject() failed

MBalazs90, 2017-01-24:
I would like to use the following code to get the weather data and parse it, then display on matrix (NodeMCU).

JSON look like:

`{"coord":{"lon":20.15,"lat":46.25},"weather":[{"id":701,"main":"Mist","description":"gyenge köd","icon":"50d"},{"id":741,"main":"Fog","description":"köd","icon":"50d"}],"base":"stations","main":{"temp":-5.86,"pressure":1025,"humidity":92,"temp_min":-7,"temp_max":-5},"visibility":1300,"wind":{"speed":2.6,"deg":330},"clouds":{"all":90},"dt":1485260100,"sys":{"type":1,"id":5732,"message":0.0097,"country":"HU","sunrise":1485238234,"sunset":1485271993},"id":715429,"name":"Szeged","cod":200}`

```
const char *weatherHost = "api.openweathermap.org";

void getWeatherData()
{
  Serial.print("connecting to "); Serial.println(weatherHost);
  if (client.connect(weatherHost, 80)) {
    client.println(String("GET /data/2.5/weather?id=") + cityID + "&units=metric&appid=" + weatherKey + weatherLang + "\r\n" +
                "Host: " + weatherHost + "\r\nUser-Agent: ArduinoWiFi/1.1\r\n" +
                "Connection: close\r\n\r\n");
  } else {
    Serial.println("connection failed");
    return;
  }
  String line;
  int repeatCounter = 0;
  while (!client.available() && repeatCounter < 10) {
    delay(500);
    Serial.println("w.");
    repeatCounter++;
  }
  while (client.connected() && client.available()) {
    char c = client.read(); 
    if (c == '[' || c == ']') c = ' ';
    line += c;
  }

  client.stop();

  DynamicJsonBuffer jsonBuf;
  JsonObject &root = jsonBuf.parseObject(line);
  if (!root.success())
  {
    Serial.println("parseObject() failed");
    return;
  }
  //weatherMain = root["weather"]["main"].as<String>();
  weatherDescription = root["weather"]["description"].as<String>();
  weatherDescription.toLowerCase();
  //  weatherLocation = root["name"].as<String>();
  //  country = root["sys"]["country"].as<String>();
  temp = root["main"]["temp"];
  humidity = root["main"]["humidity"];
  pressure = root["main"]["pressure"];
  tempMin = root["main"]["temp_min"];
  tempMax = root["main"]["temp_max"];
  windSpeed = root["wind"]["speed"];
  clouds = root["clouds"]["all"];
  String deg = String(char('~'+25));
  weatherString = "t: " + String(temp,1)+" ";
  weatherString += weatherDescription;
  weatherString += " Parat.: " + String(humidity) + "% ";
  weatherString += "Legny.: " + String(pressure/1.3332239) + " mm ";
//  weatherString += "Felhozet: " + String(clouds) + "% ";
  weatherString += "Szel: " + String(windSpeed,1) + "m/s";
}
```
I always go this error:

So connection is OK but parsing is failed.

```
connecting to api.openweathermap.org
parseObject() failed
Data loaded
```

bblanchon, 2017-01-24:
Hello Balázs,

As stated in the [FAQ](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#case-1-the-input-is-not-a-valid-json), the number reason for failing parse is a bad input.
So the first thing to do is to `Serial.println(line)` before calling `parseObject()`.

See:

* [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)
* [Examples: JsonHttpClient.ino](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonHttpClient/JsonHttpClient.ino)
* [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/)

MBalazs90, 2017-01-24:
`Serial.println(line)` output:

```
connecting to api.openweathermap.org
{"coord":{"lon":20.15,"lat":46.25},"weather": {"id":701,"main":"Mist","description":"mist","icon":"50d"},{"id":741,"main":"Fog","description":"fog","icon":"50d"} ,"base":"stations","main":{"temp":-5.84,"pressure":1025,"humidity":100,"temp_min":-7,"temp_max":-5},"visibility":1200,"wind":{"speed":3.1,"deg":330},"clouds":{"all":90},"dt":1485265500,"sys":{"type":1,"id":5732,"message":0.0916,"country":"HU","sunrise":1485238229,"sunset":1485271999},"id":715429,"name":"Szeged","cod":200}
parseObject() failed
Data loaded
```

bblanchon, 2017-01-24:
> Unexpected token { in JSON at position 105

This is because you replaced `[` and `]` with spaces.
Don't do that and call `parseArray()` instead of `parseObject()`.

MBalazs90, 2017-01-24:
ok, thanks :) you can close this

bblanchon, 2017-01-24:
You're welcome Balázs..
Thank you for using ArduinoJson.

closed 2017-01-24 by bblanchon