bblanchon/ArduinoJson #431: JSON String from https client

chris1howell, 2017-01-24:
I am attempting to decode a JSON String output from a https client on a ESP8266.

The JSON output from client.readString is 

>  
{"response":{"charging_state":"Disconnected","charge_limit_soc":90,"charge_limit_soc_std":90,"charge_limit_soc_min":50,"charge_limit_soc_max":100,"charge_to_max_range":false,"battery_heater_on":false,"not_enough_power_to_heat":false,"max_range_charge_counter":0,"fast_charger_present":false,"fast_charger_type":"<invalid>","battery_range":193.87,"est_battery_range":110.9,"ideal_battery_range":242.14,"battery_level":84,"usable_battery_level":84,"battery_current":-0.4,"charge_energy_added":0.0,"charge_miles_added_rated":0.0,"charge_miles_added_ideal":0.0,"charger_voltage":0,"charger_pilot_current":48,"charger_actual_current":0,"charger_power":0,"time_to_full_charge":0.0,"trip_charging":false,"charge_rate":0.0,"charge_port_door_open":false,"scheduled_charging_start_time":null,"scheduled_charging_pending":false,"user_charge_enable_request":null,"charge_enable_request":false,"charger_phases":null,"charge_port_latch":"Engaged","charge_current_request":48,"charge_current_request_max":48,"charge_port_led_color":"Off","managed_charging_active":false,"managed_charging_user_canceled":false,"managed_charging_start_time":null,"motorized_charge_port":true,"eu_vehicle":false,"timestamp":1485277537591}}
> 

I have tried every variation in the examples but can not seem to get the data to parse.

`   

    DynamicJsonBuffer jsonBuffer;
    String json = client.readString();
    Serial.println(json);
    JsonObject& root = jsonBuffer.parseObject(json);
    JsonObject& response = root["response"];
    const char* response_charging_state = response["charging_state"]; 
    int response_charge_limit_soc = response["charge_limit_soc"];
    int response_charge_limit_soc_std = response["charge_limit_soc_std"]; 
    int response_charge_limit_soc_min = response["charge_limit_soc_min"]; 
    int response_charge_limit_soc_max = response["charge_limit_soc_max"]; 
    bool response_charge_to_max_range = response["charge_to_max_range"]; 
    bool response_battery_heater_on = response["battery_heater_on"]; 
    bool response_not_enough_power_to_heat = response["not_enough_power_to_heat"]; 
    int response_max_range_charge_counter = response["max_range_charge_counter"]; 
    bool response_fast_charger_present = response["fast_charger_present"]; 
    const char* response_fast_charger_type = response["fast_charger_type"]; 
    float response_battery_range = response["battery_range"]; 
    float response_est_battery_range = response["est_battery_range"]; 
    float response_ideal_battery_range = response["ideal_battery_range"]; 
    int response_battery_level = response["battery_level"];
    int response_usable_battery_level = response["usable_battery_level"];
    float response_battery_current = response["battery_current"]; 
    float response_charge_energy_added = response["charge_energy_added"]; 
    float response_charge_miles_added_rated = response["charge_miles_added_rated"];
    int response_charge_miles_added_ideal = response["charge_miles_added_ideal"];
    int response_charger_voltage = response["charger_voltage"]; 
    int response_charger_pilot_current = response["charger_pilot_current"];
    int response_charger_actual_current = response["charger_actual_current"];
    int response_charger_power = response["charger_power"]; 
    int response_time_to_full_charge = response["time_to_full_charge"]; 
    bool response_trip_charging = response["trip_charging"]; 
    int response_charge_rate = response["charge_rate"]; 
    bool response_charge_port_door_open = response["charge_port_door_open"]; 
    bool response_scheduled_charging_pending = response["scheduled_charging_pending"]; 
    bool response_charge_enable_request = response["charge_enable_request"]; 
    const char* response_charge_port_latch = response["charge_port_latch"]; 
    int response_charge_current_request = response["charge_current_request"];
    int response_charge_current_request_max = response["charge_current_request_max"]; 
    const char* response_charge_port_led_color = response["charge_port_led_color"];
    bool response_managed_charging_active = response["managed_charging_active"]; 
    bool response_managed_charging_user_canceled = response["managed_charging_user_canceled"];
    bool response_motorized_charge_port = response["motorized_charge_port"]; 
    bool response_eu_vehicle = response["eu_vehicle"]; 
    long response_timestamp = response["timestamp"]; 
    Serial.print("Range - ");
    Serial.println(response_est_battery_range);
    Serial.print("SOC - ");
    Serial.println(response_battery_level);
    break;

`

bblanchon, 2017-01-24:
Hello Chris,

In your program above, I replaced `String json = client.readString();` by:

```c++
String json = R"({"response":{"charging_state":"Disconnected","charge_limit_soc":90,"charge_limit_soc_std":90,"charge_limit_soc_min":50,"charge_limit_soc_max":100,"charge_to_max_range":false,"battery_heater_on":false,"not_enough_power_to_heat":false,"max_range_charge_counter":0,"fast_charger_present":false,"fast_charger_type":"","battery_range":193.87,"est_battery_range":110.9,"ideal_battery_range":242.14,"battery_level":84,"usable_battery_level":84,"battery_current":-0.4,"charge_energy_added":0.0,"charge_miles_added_rated":0.0,"charge_miles_added_ideal":0.0,"charger_voltage":0,"charger_pilot_current":48,"charger_actual_current":0,"charger_power":0,"time_to_full_charge":0.0,"trip_charging":false,"charge_rate":0.0,"charge_port_door_open":false,"scheduled_charging_start_time":null,"scheduled_charging_pending":false,"user_charge_enable_request":null,"charge_enable_request":false,"charger_phases":null,"charge_port_latch":"Engaged","charge_current_request":48,"charge_current_request_max":48,"charge_port_led_color":"Off","managed_charging_active":false,"managed_charging_user_canceled":false,"managed_charging_start_time":null,"motorized_charge_port":true,"eu_vehicle":false,"timestamp":1485277537591}})";
```

...and it worked as expected on my ESP8266.

Make sure you don't run out of memory.

By the way, if you write `JsonObject& root = jsonBuffer.parseObject(client);`, you'll save a lot of memory.
And if you give the right size to the `DynamicJsonBuffer`, it'll be even better. Try with 2048 for instance.

See:

* [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)
* [ArduinoJson assistant](https://bblanchon.github.io/ArduinoJson/)
* [API Reference: parseObject()](https://github.com/bblanchon/ArduinoJson/wiki/API%20Reference#jsonbufferparseobject)

chris1howell, 2017-01-24:
Thank you. I was able to verify the static string.
`String json = R"({"response":{"charging_state":"Disconnected","charge_limit_soc":90,"charge_limit_soc_std":90,"charge_limit_soc_min":50,"charge_limit_soc_max":100,"charge_to_max_range":false,"battery_heater_on":false,"not_enough_power_to_heat":false,"max_range_charge_counter":0,"fast_charger_present":false,"fast_charger_type":"","battery_range":193.87,"est_battery_range":110.9,"ideal_battery_range":242.14,"battery_level":84,"usable_battery_level":84,"battery_current":-0.4,"charge_energy_added":0.0,"charge_miles_added_rated":0.0,"charge_miles_added_ideal":0.0,"charger_voltage":0,"charger_pilot_current":48,"charger_actual_current":0,"charger_power":0,"time_to_full_charge":0.0,"trip_charging":false,"charge_rate":0.0,"charge_port_door_open":false,"scheduled_charging_start_time":null,"scheduled_charging_pending":false,"user_charge_enable_request":null,"charge_enable_request":false,"charger_phases":null,"charge_port_latch":"Engaged","charge_current_request":48,"charge_current_request_max":48,"charge_port_led_color":"Off","managed_charging_active":false,"managed_charging_user_canceled":false,"managed_charging_start_time":null,"motorized_charge_port":true,"eu_vehicle":false,"timestamp":1485277537591}})";`

Works just fine.

However neither method of pulling the string from the client seem to work. 

`String json = client.readString();`
or
`JsonObject& root = jsonBuffer.parseObject(client);
`
Printing the output of json after reading from the client results in the same string... Will keep working on it....

chris1howell, 2017-01-24:
Got it.....Despite reading the FAQs and Parsing Issues. I was failing to strip the Header...

Thanks for the help!

bblanchon, 2017-01-25:
You're welcome, Chris.
Thank you for using ArduinoJson, and thank you very much for the donation 👍 

BTW, have a look at [JsonHttpClient.ino](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonHttpClient/JsonHttpClient.ino).

closed 2017-01-25 by bblanchon