bblanchon/ArduinoJson #432: ArduinoJson 5.8.2/5.8.1 & Sloeber eclipse plugin V4

hreintke, 2017-01-25:
Hi,

I do not know whether it is an issue with ArduinoJson or Sloeber eclipse plugin but using the combination I get errors for both Arduino Nano & ESP8266 projects.
``
C:/Utilities/sloeber/arduinoPlugin/libraries/ArduinoJson/5.8.2/fuzzing/fuzzer.cpp:3:39: error: expected class-name before '{' token
 class memstream : public std::istream {
``
Using 5.8.0 -> no problem

bblanchon, 2017-01-25:
Hello,

ArduinoJson is a header-only library; you're not supposed to compile any of the `.cpp` from the repository, as they are only used for internal tests.

How did you download the library?
Did you use the `.zip` or did you clone the repo?
Do you know if we can tell Sloeber to ignore certain folders?

hreintke, 2017-01-25:
Hi,
I do not know the details on it (yet). Starting to use it just now.
Using the "library manager" from it where I also can choose between the versions.

Also submitted an issue at the plugin repo. Probably an answer can com from there.

https://github.com/Sloeber/arduino-eclipse-plugin/issues/642

bblanchon, 2017-01-26:
I don't think there is anything I can do.
I added [a dedicated section on the Wiki](https://github.com/bblanchon/ArduinoJson/wiki/Compatibility-issues#sloeber-arduino-eclipse-plugin).

closed 2017-01-26 by bblanchon

hreintke, 2017-01-26:
@bblanchon : 
Thanks for your help on this.