bblanchon/ArduinoJson #435: Using as state keeper

dvv, 2017-02-02:
Hello, am looking for a way to keep a state "map" of a system. Initially empty, it is updated via incoming JSON key-value messages. I'd like to both keep the state and react to changes to the state.
Below is a draft of what I mean.

```
// FIXME 1: global (https://bblanchon.github.io/ArduinoJson/faq/why-shouldnt-i-use-a-global-jsonbuffer)
// ^^^^^^^^^^^^
static StaticJsonBuffer<1024> stateBuffer;
static JsonObject& state = stateBuffer.createObject();

// updater
void onData(char *buf)
{
    StaticJsonBuffer<256> jsonBuffer;
    JsonObject& msg = jsonBuffer.parseObject(buf);
    if (!msg.success()) return;
    for (auto kv : msg) {
      // FIXME 2: no comparison for JsonVariantBase vs JsonVariantBase
      // ^^^^^^^^^^^^
      if (kv.value != state[kv.key]) {
        state[kv.key] = kv.value;
        onChanged(kv.key, kv.value);
      }
    }
}

// reactor
void onChanged(char *key, JsonVariant& value)
{
    // custom logic, may involve global state
}

```

Questions:
- on FIXME 1: is it legal to use global here as it is only updated, not used for parsing?
- on FIXME 2: is there a method to compare two variants (other than serializing them to JSON and comparing resulting strings which is inelegant and ineffective)?
- is there an alternate solution to the problem of maintaining the state?

TIA,
--Vladimir

bblanchon, 2017-02-02:
Hello Vladimir,

Unfortunately, because of its fixed allocation strategy, ArduinoJson is not well suited for that.
In theory, you could use a global `JsonBuffer` since you're reusing the same slots instead of creating new ones.
But in practice, if your JSON contains a string, it'll be copied in the temporary `JsonBuffer` and will be invalid when you read the global `JsonObject` later on.

My advice is to use a `std::map`, a custom `struct` or another library.

I opened #436 for "FIXME 2".

dvv, 2017-02-02:
Thank you. I see. Can `std::map` map `const char *`s to `JsonVariant`s?
Couldn't you draft some lines on how it might look like?

bblanchon, 2017-02-02:
No, it cannot as the string would still be stored in the `JsonBuffer`, you need to use a `std::map<std::string,std::variant<....>>`.
This is a lot of work, I do not have the time to write it for you. Sorry.

If I were you, I would ask myself if I really need to solve such a general problem or if a simple solution is applicable.

dvv, 2017-02-02:
Thank you for the pointers!

closed 2017-02-02 by dvv

bblanchon, 2017-02-02:
You're welcome.
Thank you for opening this issue, as this gives me directions for the v6.