bblanchon/ArduinoJson #437: Help to optimize my code correctly

renat2985, 2017-02-02:
Please help to correctly optimize my code for ArduinoJson

JSON format:
```
[{"ssid":"Laura","dbm":-81,"pass":"*"},{"ssid":"www.etaxi.lv","dbm":-48,"pass":"*"},{"ssid":"P110006"
,"dbm":-85,"pass":"*"},{"ssid":"WiFi_Tikls","dbm":-82,"pass":"*"}]
```

My code:
 ``` int n = WiFi.scanNetworks();
  String wifiScan = "[";
  if (n == 0)
    wifiScan = "{\"ssid\":\"none\"}";
  else {
    for (int i = 0; i < n - 1; ++i) {
      wifiScan += "{";
      wifiScan += "\"ssid\":\"";
      wifiScan += WiFi.SSID(i);
      wifiScan += "\",";
      wifiScan += "\"dbm\":";
      wifiScan += WiFi.RSSI(i);
      wifiScan += ",";
      wifiScan += "\"pass\":\"";
      wifiScan += (WiFi.encryptionType(i) == ENC_TYPE_NONE) ? "" : "*";
      wifiScan += "\"}";
      if (i != n - 2) wifiScan += ",";
      delay(10);
    }
    wifiScan += "]";
  }
  HTTP.send(200, "text/json", wifiScan);

renat2985, 2017-02-03:
Thanks
```
  int n = WiFi.scanNetworks();
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  JsonArray& networks = root.createNestedArray("networks");
  for(int i=0;i<n;i++) {
    JsonObject& network = networks.createNestedObject();
    network["ssid"] = WiFi.SSID(i);
    network["pass"] = (WiFi.encryptionType(i) == ENC_TYPE_NONE) ? "" : "*";
    network["dbm"] = WiFi.RSSI(i);
    network["bssid"] = WiFi.BSSIDstr(i);
    network["channel"] = WiFi.channel(i);
    network["isHidden"] = WiFi.isHidden(i);
  }
  String retJson;
  root.printTo(retJson);
  HTTP.send(200, "text/json", retJson);
```

bblanchon, 2017-02-03:
Hello Renat,

I'm glad you found the answer you were looking for.

I don't know what class is `HTTP`, but if it implements the `Print` interface, then you can do:

```
retJson.printTo(HTTP);
```

Just after sending the HTTP headers.
It would save a useless copy of the data.

Also, you should provide an initial size of the DynamicJsonBuffer.
You can use [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/) to find the right size.

renat2985, 2017-02-03:
Yes I use `#include <ESP8266WebServer.h>`
I have an error. What do you think, what's the problem?
[image]

bblanchon, 2017-02-03:
Nevermind, it's not possible to do that with ESP8266WebServer.
Anyway, there are so many dynamic allocation and string duplications in ESP8266WebServer that I'm not sure it's worth optimizing the JSON side.

closed 2017-02-04 by bblanchon

gabrielklein, 2017-04-15:
Instead of json.printTo(HTTP);        

Do
String s;
json.printTo(s);
this->webServer->send(200, "text/json", s);

ESP8266 contains 4Mo of memory.

Where webServer is 
ESP8266WebServer *webServer =  new ESP8266WebServer(80);

bblanchon, 2017-04-15:
@gabrielklein According to [wikipedia](https://en.wikipedia.org/wiki/ESP8266) the ESP8266 has 96KB of RAM for data.