bblanchon/ArduinoJson #438: Encoding Json Error

fernandomorse, 2017-02-02:
Hello,
I need to encode a json that has a value that is an array of arrays, like this:

```
{
	"resources": [{
		"name": "Rele0",
		"rsc": []
	}, {
		"name": "Rele1",
		"rsc": []
	}, {
		"name": "Rele2",
		"rsc": []
	}, {
		"name": "Dht0",
		"rsc": ["Umidade", "Temperatura"]
	}]
}
```
My code is 

      ```
           JsonObjectStream* stream = new JsonObjectStream();
		JsonObject& root = stream->getRoot();
		JsonArray& data = root.createNestedArray("resources");

		for (int i = 0; i <=_indiceJSON-1; i++)
		{
			JsonObject& item = data.createNestedObject();
			item["name"] = _resourcesJSON[i];
			
			JsonArray& rsc = root.createNestedArray("rsc");	

			rsc.add(_subresourcesJSON[i][0]);
			rsc.add(_subresourcesJSON[i][1]);
		}

```
Any idea ???

bblanchon, 2017-02-03:
Hello @fernandomorse 

I bet that the `JsonBuffer` is too small.

See:

* [FAQ: Why some parts are missing?](https://bblanchon.github.io/ArduinoJson/faq/why-some-parts-are-missing)

fernandomorse, 2017-02-03:
@bblanchon Hello,

Thanks for the answer, StaticJsonBuffer is the right size, however my problem is as follows, I need inside the data object to create the array "rsc" and add values to the array "rsc" (rsc.add (value)), And I'm not getting it, do you have any examples?

Structure of JSON

```
Root -> object
Resource -> root array
Date -> resource object
Rsc -> resource array
           Rsc.add ()
```

I can not serialize rsc.

Thanks

bblanchon, 2017-02-03:
Is it possible that `_subresourcesJSON` goes out of scope before you call `root.printTo()`?

See:

* [Avoiding pitfalls: Do not assume that strings are copied](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#6-do-not-assume-that-strings-are-copied)

fernandomorse, 2017-02-03:
Thanks @bblanchon 

Problem resolved.

Please Close  issue.

bblanchon, 2017-02-04:
What was the solution?

fernandomorse, 2017-02-04:
Error lovical.  Variable._inxiceJSON in the loop for,  was wrong,  value in the example correct 4 but value iwas 6. No problem in JSON.

closed 2017-02-05 by bblanchon