bblanchon/ArduinoJson #442: how to use reference "root" as a parameter in a function?

tarzan115, 2017-02-08:
there is my question.
I have a function like:
```c
void parseJson_sub(ServerDATA server_Data_sub, JsonObject& root_sub)
{
   server_Data_sub.Temperature = root_sub["TEMP"];
}
```
and an other function call `parseJson_sub()` function like:
```c
bool parseJson(char* json)
{
   ServerDATA server_Data;
   StaticJsonBuffer<SERVER_DATA_JSON_SIZE> jsonBuffer;
   JsonObject& root = jsonBuffer.parseObject(json);

   if(!root.success())
   {
    PRINT_LOG("parseObject() failed");
    return false;
   }
   server_Data.Time = root["TIME"];
   parseJson_sub(server_Data, root);
}
```

but I got the error, how to use `root` as a parameter for a function?

thank you so much.
have a good day 😃

bblanchon, 2017-02-08:
Hi Doanh,
Your code looks OK.
What error do you get?

tarzan115, 2017-02-08:
ohh, I'm sorry.
in the first time, I got an error. so I thought I'm doing wrong way.
thank you for your support. 😃

closed 2017-02-08 by tarzan115