bblanchon/ArduinoJson #443: Json not working on Nano

tbarritt, 2017-02-08:
I used the following to parse data on ESP8266 12 and it worked fine. Then tried on nano, lock up after it tries to parseObject. Below is just the parsing part and serial print out.

```ccp
  Serial.println("WebHook data:");
  Serial.println(param.asStr());
  
  const size_t bufferSize = JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(10) + 490;
  
  DynamicJsonBuffer jsonBuffer(bufferSize);
     
  JsonObject& root = jsonBuffer.parseObject(param.asStr());
    
  // Test if parsing succeeds.
  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }

  JsonObject& results = root["results"];
  const char* results_sunrise = results["sunrise"]; // "2017-02-03T13:24:21+00:00"
  const char* results_sunset = results["sunset"]; // "2017-02-03T23:45:39+00:00"
  const char* results_solar_noon = results["solar_noon"]; // "2017-02-03T18:35:00+00:00"
  long results_day_length = results["day_length"]; // 37278
  const char* results_civil_twilight_begin = results["civil_twilight_begin"]; // "2017-02-03T12:56:31+00:00"
  const char* results_civil_twilight_end = results["civil_twilight_end"]; // "2017-02-04T00:13:29+00:00"
  const char* results_nautical_twilight_begin = results["nautical_twilight_begin"]; // "2017-02-03T12:24:48+00:00"
  const char* results_nautical_twilight_end = results["nautical_twilight_end"]; // "2017-02-04T00:45:12+00:00"
  const char* results_astronomical_twilight_begin = results["astronomical_twilight_begin"]; // "2017-02-03T11:53:35+00:00"
  const char* results_astronomical_twilight_end = results["astronomical_twilight_end"]; // "2017-02-04T01:16:25+00:00"
  
  const char* status = root["status"]; // "OK"
  
  // Print values.
  Serial.print("sunrise\t\t"); Serial.println(results_sunrise);
  Serial.print("sunset\t\t"); Serial.println(results_sunset);
  Serial.print("day length\t");  Serial.println(results_day_length);
```

WebHook data:
{"results":{"sunrise":"2017-02-08T13:19:09+00:00","sunset":"2017-02-08T23:51:25+00:00","solar_noon":"2017-02-08T18:35:17+00:00","day_length":37936,"civil_twilight_begin":"2017-02-08T12:51:38+00:00","civil_twilight_end":"2017-02-09T00:18:57+00:00","nautical_twilight_begin":"2017-02-08T12:20:09+00:00","nautical_twilight_end":"2017-02-09T00:50:26+00:00","astronomical_twilight_begin":"2017-02-08T11:49:05+00:00","astronomical_twilight_end":"2017-02-09T01:21:30+00:00"},"status":"OK"}
ical_twilight_end":"20

tbarritt, 2017-02-08:
If I can't get the nano to work, how can I parse the results?
```ccp
2017-02-08T13:19:09+00:00```

bblanchon, 2017-02-08:
`bufferSize` equals `618` on the nano.
This device only has 2KB or RAM, so this is very tight.

To optimize memory usage, use a `char[]` to hold the JSON and use a `StaticJsonBuffer`.
The buffer can be much smaller since you don't need to duplicate the input.

Also, make sure that no other big object stays in memory while the JSON is in memory.

See:

* [FAQ: What are the differences between StaticJsonBuffer and DynamicJsonBuffer?](https://bblanchon.github.io/ArduinoJson/faq/what-are-the-differences-between-staticjsonbuffer-and-dynamicjsonbuffer)
* [API Reference: JsonBuffer::parseObject()](https://bblanchon.github.io/ArduinoJson/api/jsonbuffer/parseobject/)

tbarritt, 2017-02-08:
When I change to;
```ccp
  char data [] = (param.asStr());
  
  StaticJsonBuffer<700> jsonBuffer;
     
  JsonObject& root = jsonBuffer.parseObject(data);
```
I get initializer fails to determine size of 'data'

tbarritt, 2017-02-08:
I changed to char* and it worked.

Any easy way to parse out?      2017-02-08T13:19:09+00:00

Thank you for the help.

bblanchon, 2017-02-08:
```c++
const size_t bufferSize = JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(10);
JsonObject& root = jsonBuffer.parseObject((char*)param.asStr());
```

This can be a very simple solution indeed :-)
Be aware that the content of `param` is changed by `parseObject()`.

tbarritt, 2017-02-08:
Any way to parse out? 2017-02-08T13:19:09+00:00

bblanchon, 2017-02-08:
Try [sscanf](http://www.cplusplus.com/reference/cstdio/sscanf/).

tbarritt, 2017-02-08:
I have tried it, it errors and reconnects.

```ccp
  sscanf(results_sunset,"%4d- %2d- %2dT %2d: %2d: %2d+",&YY, &MM ,&DD, &hh, &mm, &ss);
  
  Serial.print("hour:\t");   Serial.println(hh);
  Serial.print("minute:\t"); Serial.println(mm);
  Serial.print("second:\t"); Serial.println(ss);
  Serial.print("month:\t");  Serial.println(MM);
  Serial.print("day:\t");    Serial.println(DD);
  Serial.print("year:\t");   Serial.println(YY);
```
```ccp
  ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.4.4 on Arduino Nano

[5001] Connecting to blynk-cloud.com:8442
[5401] Ready (ping: 87ms).
WebHook data:
{"results":{"sunrise":"2017-02-08T13:19:09+00:00","sunset":"2017-02-08T23:51:25+00:00","solar_noon":"2017-02-08T18:35:17+00:00","day_length":37936,"civil_twilight_begin":"2017-02-08T12:51:38+00:00","civil_twilight_end":"2017-02-09T00:18:57+00:00","nautical_twilight_begin":"2017-02-08T12:20:09+00:00","nautical_twilight_end":"2017-02-09T00:50:26+00:00","astronomical_twilight_begin":"2017-02-08T11:49:05+00:00","astronomical_twilight_end":"2017-02-09T01:21:30+00:00"},"status":"OK"}
sunrise		2017-02-08T13:19:09+00:00
sunset		2017-02-08T23:51:25+00:00
day length	37936
h[1] Getting IP...
[1816] IP:192.168.1.30
[1816] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.4.4 on Arduino Nano
```

tbarritt, 2017-02-08:
ok close this time with
```ccp
  int sscanf(results_sunset,"%4d- %2d- %2dT %2d: %2d: %2d+",&YY, &MM ,&DD, &hh, &mm, &ss);
```
```ccp
WebHook data:
{"results":{"sunrise":"2017-02-08T13:19:09+00:00","sunset":"2017-02-08T23:51:25+00:00","solar_noon":"2017-02-08T18:35:17+00:00","day_length":37936,"civil_twilight_begin":"2017-02-08T12:51:38+00:00","civil_twilight_end":"2017-02-09T00:18:57+00:00","nautical_twilight_begin":"2017-02-08T12:20:09+00:00","nautical_twilight_end":"2017-02-09T00:50:26+00:00","astronomical_twilight_begin":"2017-02-08T11:49:05+00:00","astronomical_twilight_end":"2017-02-09T01:21:30+00:00"},"status":"OK"}
sunrise		2017-02-08T13:19:09+00:00
sunset		2017-02-08T23:51:25+00:00
day length	37936
hour:	0
minute:	0
second:	0
month:	0
day:	0
year:	0
```

bblanchon, 2017-02-08:
```c++
#include <stdio.h>

int main()
{
  const char * input = "2017-02-08T23:51:25+00:00";

  int year, month, day, hour, minute, second;

  sscanf(input, "%d-%d-%dT%d:%d:%d", &year, &month, &day, &hour, &minute, &second);

  printf("year = %d\n", year);
  printf("month = %d\n", month);
  printf("day = %d\n", day);
  printf("hour = %d\n", hour);
  printf("minute = %d\n", minute);
  printf("second = %d\n", second);
}
```

tbarritt, 2017-02-08:
your way through me offline and starts over. My way was close but got 0 for all results.

tbarritt, 2017-02-08:
using just **sscanf** throws me offline and starts over, **int sscanf** gives all zeros for results.

tbarritt, 2017-02-08:
got it to work,,,

thanks again for all the help.

bblanchon, 2017-02-08:
You're welcome @tbarritt, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2017-02-08 by bblanchon