bblanchon/ArduinoJson #445: Socket will not be closed when Connection failes

Chris685, 2017-02-10:
At the Arduino Example the Network sockets will not be closed when the Connection could not be established. After 4 trys - my WiFi stopps working and could only be started to work.

My workaround was to put the disconnect directly to the loop, so it will called every loop - also if succesfull or unsuccesfull.

Original Code from Example:
`// ARDUINO entry point #2: runs over and over again forever
void loop() {
  if (connect(server)) {
    if (sendRequest(server, resource) && skipResponseHeaders()) {
      UserData userData;
      if (readReponseContent(&userData)) {
        printUserData(&userData);
      }
    }
    **disconnect();**
  }
  wait();
}
`

My Workaround:
`
void loop() {
  if (connect(server)) {
    if (sendRequest(server, resource) && skipResponseHeaders()) {
      UserData userData;
      if (readReponseContent(&userData)) {
        printUserData(&userData);
      }
    }
  }
  **disconnect();**
  wait();
}
`

bblanchon, 2017-02-10:
Hello Chris,

I'm quite surprised we need to disconnect when the connection failed.
Can you confirm that this workaround solves your problem?
If so, can you make a Pull Request?

Chris685, 2017-02-10:
My Problem was solved:

#include <utility/server_drv.h>   //WIFI Sockets

    for(int x = 0; x < MAX_SOCK_NUM; x++) {
    Serial.print(WiFi._state[x]);   
    Serial.print("  ");
    Serial.print(WiFi._server_port[x]);   
    Serial.print("  s=");
    Serial.print(serverDrv.getServerState(x));   
    Serial.print("  c=");
    Serial.print(serverDrv.getClientState(x));   
    Serial.print("  d=");
    Serial.println(serverDrv.availData(x));
    }

I watched my sockets - so i believe it solves my problem. But i dont know if it is a good way - probably its also possible to close the socket directly by the command here:

bool connect(const char* cNexadresse) {
  Serial.print("Verbinde zu ");
  Serial.println(cNexadresse);
  bool ok = client.connect(cNexadresse, 80);
  Serial.println(ok ? "Verbunden mit Resol DL2" : "Verbindung fehlgeschlagen!");
  return ok;
}

Your code is a little bit adjusted to fit to my project

bblanchon, 2017-02-11:
I merge you PR.
Thank you very much Chris 👍

closed 2017-02-11 by bblanchon