bblanchon/ArduinoJson #449: I could not clear jsonbuffer.

Joypacharya, 2017-02-13:
I want to flush json buffer, but i faced some issue regarding this factor.
I did try some trick still my ESP had been reset after some time.

One time i used 
StaticJsonBuffer<5000>jsonBuffer; and tested for that 

Even i used
DynamicJsonBuffer jsonBuffer;
blah...blah.
{
}
jsonBuffer = 0; // Reset the value for Jsonbuffer. I am not sure about this.So, i tried it.

So,  please help me for this problem.Give me proper steps for this problem.Thank you.

bblanchon, 2017-02-13:
Hello Joy,

See:

* [FAQ: Why does my device crash or reboot?](https://bblanchon.github.io/ArduinoJson/faq/why-does-my-device-crash-or-reboot)
* [FAQ: Why shouldn't I use a global JsonBuffer?](https://bblanchon.github.io/ArduinoJson/faq/why-shouldnt-i-use-a-global-jsonbuffer)
* [FAQ: What's the best way to use the library?](https://bblanchon.github.io/ArduinoJson/faq/whats-the-best-way-to-use-the-library)
* [FAQ: How to reuse a JsonBuffer?](https://bblanchon.github.io/ArduinoJson/faq/how-to-reuse-a-jsonbuffer)

I believed this was a well-covered topic in the documentation; I'm quite surprised you didn't find it.
Do you have any suggestion to improve the documentation?

Joypacharya, 2017-02-14:
If i am using this syntax  "jsonBuffer.clear();" then it display this kind of error. So,help me resolve this problem,
 
RX:200: error: 'ArduinoJson::DynamicJsonBuffer' has no member named 'clear'

       jsonBuffer.clear();

                  ^
'ArduinoJson::DynamicJsonBuffer' has no member named 'clear'

"Display shows this type of error."

bblanchon, 2017-02-14:
Hi Joy,

I'm sorry, there is a misunderstanding: there is no `.clear()` method.
You can [write your own](https://github.com/bblanchon/ArduinoJson/wiki/Bag-of-Tricks#reuse-jsonbuffer), but it's a recipe for failure.
Please read the documentation and the FAQ.

Joypacharya, 2017-02-14:
Ok ..but can you tell me where i put this thing in my code any example for that.
I am beginner.Currently i can not place as per suggestion.So, help me resolve this point.

bblanchon, 2017-02-14:
I'm sorry but I cannot help you since you gave me close to zero information on your project.

However, all the information you need is available :-)
Read the [doc](https://github.com/bblanchon/ArduinoJson/wiki), read the [examples](https://github.com/bblanchon/ArduinoJson/tree/master/examples), and look at [existing projects](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Joypacharya, 2017-02-14:
i am sending and receive inform via JSON string.I parses object limited time after that i cant parse any object.So, help for parses object easily without stack problem or overflow.I am using MQTT for subscribe and publish message.IS their any reason for that one?
I am not sure about that one.Not possible to give code or paste code in this conversational space.And thanks for helping my problem.

bblanchon, 2017-02-14:
See:

* [FAQ: The first parsing succeeds, why do the next ones fail?](https://bblanchon.github.io/ArduinoJson/faq/the-first-parsing-succeeds-why-do-the-next-ones-fail)

closed 2017-02-19 by bblanchon