bblanchon/ArduinoJson #450: How to get array size?

nnvn, 2017-02-13:
How to get size of *myArray* in
{
    "x": "y",
    "myArray": [ "a", "b", "c" ]
}

bblanchon, 2017-02-13:
Hello @nnvn,

[JsonArray::size()](https://bblanchon.github.io/ArduinoJson/api/jsonarray/size/) returns the number of elements in the array.

I believed this was a well-covered topic in the documentation; I'm quite surprised you didn't find it.
Do you have any suggestion to improve the documentation?

nnvn, 2017-02-13:
Hi, 
```
  char json[]="{ \"x\": \"y\",  \"myArray\": [ \"a\", \"b\", \"c\" ] };
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(json);
  for (int i=0; i < ??????; i++){
       Serial.println((const char*)(root["myArray"][0]));
  } 
 
```

bblanchon, 2017-02-13:
To read all elements of an array, you should use [iterators](https://bblanchon.github.io/ArduinoJson/api/jsonarray/begin_end/):

```c++
JsonArray& myArray = root["myArray"];
for (JsonVariant myValue : myArray) {
  Serial.println(myValue.as<char*>());
} 
```

nnvn, 2017-02-14:
Great. It worked. Thank you. 

I got confused initially because *myArray* is actually an array of objects. I had looked at examples, bagOfTricks etc, but couldn't figure it out. With your hint I made *JsonObject & t    =   (myArray.get<JsonVariant>(i)).as<JsonObject&>();* and it's working.. 

Also, Thanks for the awesome work.

closed 2017-02-14 by nnvn

bblanchon, 2017-02-15:
Hi @nnvn,

Your code will be faster if you use iterators as suggested in my message.

Here is how I would rewrite this piece of code:

```c++
JsonArray& scheduleArray = root["Schedule"];
for (JsonObject& t : scheduleArray) {

}
```

nnvn, 2017-02-25:
Hi, did it. works. Thanks.