bblanchon/ArduinoJson #451: JSON parsing failed!

RamanujamV, 2017-02-14:
Setup:
NodeMCU 1.0 (ESP-12E) Module (LoLin v3) Connected via USB port to Raspberry Pi 3 running Arduino IDE 1.8.2
The Sketch used and the Error Message as seen in the Serial Monitor (and copied) are in the files attached.
The Json received from the Nominatim Server has been verified to have correct syntax at 
http://www.jsoneditoronline.org/  

Thanking you for your help in understanding /resolving this error.
- RamanujamV

bblanchon, 2017-02-14:
Hi @RamanujamV,

See:

* [FAQ: Why parsing fails?](https://bblanchon.github.io/ArduinoJson/faq/why-parsing-fails)
* [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/)

RamanujamV, 2017-02-15:
Hello  bblanchon,First of all Thank You very much for your prompt response.
As suggested, I modified my sketch for Memory Allocation as given under "Parsing Program" of the  web page.https://bblanchon.github.io/ArduinoJson/assistant/ I do not see any difference in the results of the sketch execution.

As for following up on Why Parsing Fails cited under he FAQ ...The parsing functions, parseArray() and parseObject(), may fail for 5 reasons:   
   - The input is not a valid JSON - The input json has been verified to be syntactically Correct
   - The StaticJsonBuffer is too small - I am using DynamicjsonBuffer to allocate memory
   - The StaticJsonBuffer is too big (stack overflow) - (see above)
   - The DynamicJsonBuffer fails to allocate memory - I confess that I really do not know how to check to see if the DynamicjsonBuffer allocated memory appropriately
   - You called parseObject() instead of parseArray() - I am indeed using parseobject in the sketch
Obviously some thing else is wrong in my sketch tat I need to fix.Thanks again for your time and assistance in responding to my query.- RamanujamV
P.S.  By the way, while seeing your (first) name, I couldn't help but think of the famous "Dr. Fractal - Benoit Mandelbrot" who was an IBM Fellow when I worked there years ago!      From: Benoît Blanchon <notifications@github.com>
 To: bblanchon/ArduinoJson <ArduinoJson@noreply.github.com> 
Cc: RamanujamV <alagar_venkataswamy@yahoo.com>; Mention <mention@noreply.github.com>
 Sent: Tuesday, February 14, 2017 2:18 PM
 Subject: Re: [bblanchon/ArduinoJson] JSON parsing failed! (#451)
   
Hi @RamanujamV,See:   
   - FAQ: Why parsing fails?
   - ArduinoJson Assistant

bblanchon, 2017-02-15:
@RamanujamV, I guess you have no choice but showing a little of your program...

RamanujamV, 2017-02-15:
I thought I attached a copy of my sketch when I reported the issue.

RamanujamV, 2017-02-15:
I just checked the GitHub site, and I do not see my attachments.
I will try to attach them again.
Thanks again.
- RamanujamV

RamanujamV, 2017-02-15:
Attaching copy of Sketch and Error message as seen in the Serial Monitor
[Json_Parser_Error_Msg.txt](https://github.com/bblanchon/ArduinoJson/files/778171/Json_Parser_Error_Msg.txt)
[Json_Parser_Sketch.txt](https://github.com/bblanchon/ArduinoJson/files/778173/Json_Parser_Sketch.txt)

Copy of  Message in the Serial Monitor (Copied here also):
Connected to wifi with IP xxx.xxx.x.x

[Connecting to nominatim.openstreetmap.org ... connected]
Request sent - waiting for reply...
JSON parsing failed!

Exception (28):
epc1=0x4000bf0e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffef400 end: 3ffef660 offset: 01a0

>>>stack>>>
3ffef5a0:  3ffe834c 3ffef5fc 3ffee600 4020248f  
3ffef5b0:  3ffef614 3ffef5fc 00000032 3ffef614  
3ffef5c0:  feefef0a feefeffe feefeffe feefeffe  
3ffef5d0:  feefeffe feefeffe feefeffe feefeffe  
3ffef5e0:  feefeffe feefeffe feefeffe feefeffe  
3ffef5f0:  feefeffe feefeffe feefeffe 3ffe8640  
3ffef600:  00000000 00002710 00005465 00000000  
3ffef610:  3fff0b3c 3ffe8628 00000000 3fff0d7c  
3ffef620:  000006dc 0a0d0a0d feefef00 feefeffe  
3ffef630:  feefeffe feefeffe feefeffe 3ffee62c  
3ffef640:  3fffdad0 00000000 3ffee624 40203964  
3ffef650:  feefeffe feefeffe 3ffee640 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

RamanujamV, 2017-02-15:
Closed by Mistake - still a "newbie" for GitHub!

bblanchon, 2017-02-15:
You specified HTTP 1.1, and therefore the server replies with 

```
Transfer-Encoding: chunked
```

This means that the JSON is preceded by an integer specifying the size of the chunk.

Switch to HTTP 1.0 and it should work.

RamanujamV, 2017-02-15:
Benoit Blanchon,That fixed the problem.Thank you very much, indeed!- RamanaujamV

closed 2017-02-19 by bblanchon

RamanujamV, 2017-02-19:
Benoit,
Thanks again for your help.

(I am not familiar with the formalities of GitHub to be sure if I should have been the one to close the issue/problem that I opened.)

Thanks again.
- RamanujamV

bblanchon, 2017-02-19:
You're welcome RamanujamV.
Either you or I can close the issue; I don't think there is an official convention.
You made it clear in your last message that your problem was solved, so that's good for me.

If you want to help me back, please click on the star at the top, this is very rewarding.

RamanujamV, 2017-02-20:
Benoit,I just Clicked on the Star at the top of the issue page.
Thanks for your tool; it is very useful.- RamanujamV