bblanchon/ArduinoJson #452: Decoding complex nested object in Array

Joypacharya, 2017-02-15:
I would like to decoding this one,
{
    "service": active,
    "data": [
        {"Code": 01,"Message": "OK"},
         {"Code": 02,"Message": "Failure"}]
}

bblanchon, 2017-02-15:
Please use [ArduinoJson assistant](https://bblanchon.github.io/ArduinoJson/assistant/).

BTW, your JSON input is not valid as it lacks quotes around strings.
ArduinoJson will parse it correctly but the Assistant will reject it.

Joypacharya, 2017-02-16:
Thanks for short out my issues.
Please give your email id.So, i will contact you directly related this topic.
Still i have issues.if possible than i will tell you.
Once again thanks a lot for help me.

bblanchon, 2017-02-16:
I'm sorry I don't give my email for that kind of things.
I'm sure you'll understand.

Can I close this issue?
What about #449?

Joypacharya, 2017-02-18:
Even i went your site arduinojson and as per recommendation. 
Still i faced issues regrading paresesobject failed();
So, help me out this problem. 
generally i am using both separated jsonbuffer for send command and receive command.
So,Is their any possibility for jsonobject failed(); 
Tell about this one if i am wrong.
I already did read your documents of wiki GitHub.Thank you.

bblanchon, 2017-02-19:
Hello Joy,
How am I supposed to help you if you don't provide any code?

DarkBeardRUS, 2017-02-21:
**Hello,
I also experienced a problem with nested object in Array.
newjsonConfig  - It contains such a conclusion
(wrote here is not complete json output, but it is definitely valid.)**

{
  "systemName": "Light",
  "systemVersion": "1.6",
  "systemConfigs": {
    "modules": [
      {
        "module position": 1,
        "count": 30,
        "role": "none"
      },
      {
        "module position": 2,
        "count": 0,
        "role": "none"
      },

**trying to parse this output is design**

  JsonObject& root = newjsonBuffer.parseObject(newjsonConfig);
  JsonObject& root_systemConfigs = root["systemConfigs"];
  JsonArray& modulesArray = root_systemConfigs["modules"];
  int m1 = modulesArray["module position"][0];

**I receive an error**

New_Config:60: error: invalid conversion from 'const char*' to 'size_t {aka unsigned int}' [-fpermissive]
 int m1 = modulesArray["module position"][0];
-------------------------------------------^
**I do not know how to fix it ...
Maybe you have thought?**

Joypacharya, 2017-02-22:
Here your solution..
I thought you do not done proper formation.
Go throw it and you will get result.
Example based i put "System config" value.You may change it.
{
"systemName": "Light",
"systemVersion": "1.6",
"systemConfigs":"Start",
"modules": [
{
"module position": 1,
"count": 30,
"role": "none"
},
{
"module position": 2,
"count": 0,
"role": "none"
}
]
}

DarkBeardRUS, 2017-02-22:
No, this will not work, it turns into a peer JSON.
http://www.jsoneditoronline.org/?id=3ccb6922530cfbf9cfb524d842f8bbbd 
the link is available for the full version of the configuration that I decoded.

DarkBeardRUS, 2017-02-22:
**I found a bug!**
int m1 = modulesArray["module position"][0]; - does not work
int m1 = modulesArray[0]["module position"]; - working

bblanchon, 2017-02-22:
@DarkBeardRUS, as you just figured out the solution is to replace:

```c++
modulesArray["module position"][0];
```
 by

```c++
modulesArray[0]["module position"];
```

This is because `modulesArray` is an array and not an object.

(Note: it's better to [open a new issue](https://github.com/bblanchon/ArduinoJson/issues/new) when you have a question, because you completely hijacked this unrelated thread.)

larsgk, 2017-02-22:
The working example also matches with the JSON...Did you mean a bug in your
code or in the lib?

bblanchon, 2017-02-28:
@Joypacharya It's been 11 days without an update from your side.
I'm closing this issue.
Reopen if you need and **include code sample**.

closed 2017-02-28 by bblanchon