bblanchon/ArduinoJson #453: Wrong deserialisation of float values with exponential notation

moodgorning, 2017-02-19:
Hi again,
I'm experiencing trouble deserialising floating point values. If I am serialising values bigger than 1000 they get encoded in the "1.0e3" notation, but if I read them from the serialised json, it ignores the exponent and reads it as 1.0.
See this example code, 11000.13465 gets serialised to 1.100013e4 but read back as 1.100013:
```
#include <ArduinoJson.h>

void setup()
{
}

float value = 11000.13465;

void readJson(String json){
	const size_t BUFFER_SIZE = JSON_OBJECT_SIZE(1) + 190;
    StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;
    JsonObject& root = jsonBuffer.parseObject(json);

    if (!root.success())
    {
      Serial.println("parseObject() failed ");
      Serial.println(jsonBuffer.size());
      Serial.println(root.success());
      return;
    }
    if(root.containsKey("value")){
      value = root["value"].as<float>();
    }else{
      Serial.println("this should never happen");
    }
    Serial.println("value: " + String(value));
}

String serializeValues(){
    const size_t bufferSize =  JSON_OBJECT_SIZE(1) + 190;
    StaticJsonBuffer<bufferSize> jsonBuffer;
    JsonObject& root = jsonBuffer.createObject();
    root["value"] = double_with_n_digits(value,6);
    String json;
    root.printTo(json);
    Serial.println(json);
   return json;
}

void loop()
{
	readJson(serializeValues());
	delay(2000);
}
```
It doesn't matter whether I use as<float>() or not. I'm running this on an esp8266-12F using platformIo and the arduino esp8266 SDK
Any help is much appreciated
Cheers
Max

bblanchon, 2017-02-19:
Hi Max,

I'm using [`strtod()`](https://github.com/bblanchon/ArduinoJson/blob/master/include/ArduinoJson/Data/Parse.hpp#L24) to convert strings to `double`.
The standard `strtod()` [works as expected](http://melpon.org/wandbox/permlink/mlJDp50WNZWKy4yc), but I bet that the implementation on ESP8266 is wrong.
I believe it's time to write my own implementation 😢 

Meanwhile, as a temporary workaround, I suggest that you change the threshold for the scientific notation.
It's on [include/ArduinoJson/Serialization/JsonWriter.hpp, line 95](https://github.com/bblanchon/ArduinoJson/blob/master/include/ArduinoJson/Serialization/JsonWriter.hpp#L95).

Thanks a lot for reporting this bug.

moodgorning, 2017-02-19:
Ah bugger that, looks like you are right, the esp8266 implementation of strtod() doesn't seem to take the scientific notation into account.

moodgorning, 2017-02-19:
These guys seemed to have the same problem, maybe their implementation atof_e is worth a look: https://digistump.com/board/index.php?topic=2531.0

closed 2017-03-19 by bblanchon