bblanchon/ArduinoJson #455: Json with nrf24l01 - Decoding Problem

gustavogini, 2017-02-28:
Hello Benoit,

Im trying to commuticate two arduino boards with nrf24l01 and facing some problems to decoding Json, I ve looked all faq`s and issues to find this kind of problem, but without success. 

I`ve calculated the buffer size and encode the message without problems.

Do you have any suggestions?

Thanks in advance! 

```
unsigned char data;
   if ( radio.available()) {
    while (radio.available()) {
      radio.read( &data, sizeof(unsigned char));
    }
     StaticJsonBuffer<200> jsonBuffer;
     JsonObject& root = jsonBuffer.parseObject(&data);
    if (!root.success()){
      Serial.println("parseObject() failed");
      return;
     }  
      int Poco01 = root["Poco01"];
      int Poco02 = root["Poco02"];
      int Rele01 = root["Rele01"];
      int Rele02 = root["Rele02"];

    Serial.println("Received Message: ");
    Serial.println(Poco01); //1
    Serial.println(Poco02); //2
    Serial.println(Rele01); //3
    Serial.println(Rele02); //4
```

bblanchon, 2017-02-28:
Your buffer is only one byte :-)

Try this:

```c++
if ( radio.available()) {
  unsigned char data[1024];
  while (radio.available()) {
    radio.read(data, sizeof(data));
  }
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(data);
```

But if `radio` implements `Stream`, you can do that:

```c++
if ( radio.available()) {
     StaticJsonBuffer<1024> jsonBuffer;
     JsonObject& root = jsonBuffer.parseObject(radio);
```

Note that I increased the size of the `JsonBuffer`, since it now holds the data.

gustavogini, 2017-02-28:
Hello, Benoit!

This is not a Stream. I implemented the first option you suggest but the `parseObject() failed`.
That`s the encoding code below. Do you see some problem on it to generate a problem on decoding?

Thanks!

```
void loop() {
  int poco01 = 1;
  int poco02 = 2;
  int rele01 = 3;
  int rele02 = 4;

StaticJsonBuffer<200> jsonBuffer;
JsonObject& root = jsonBuffer.createObject();

root["Poco01"] = poco01;
root["Poco02"] = poco02;
root["Rele01"] = rele01;
root["Rele02"] = rele02;

char buffer[256];
root.printTo(buffer, sizeof(buffer));
  char *data = buffer;
  if (!radio.write( &data, sizeof( char) )) {
    Serial.println("No acknowledgement of transmission - receiving radio device connected?");    
  }
```

bblanchon, 2017-03-01:
Hello @gustavogini,

Here again, you write only one byte.

Try:

```c++
int n = root.printTo(buffer, sizeof(buffer));
radio.write(buffer, n);
```

If `radio` implements `Print`, you can do:

```c++
root.printTo(radio);
```

If the parsing still fails, please `Serial.println()` the content of the JSON input.
Make sure the transmission works and the JSON input is valid.

gustavogini, 2017-03-01:
Hello @bblanchon 

I tested the transmission (it works) and use `Serial.println()` to check the content and it works very well too.
Im using both codes as you suggested and the decoding still `failed` but the decoding arduino IDE shows this message:

```
large implicity truncated to unsigned type [-Woverflow] 
radio.write (data, sizeof (data));
```

bblanchon, 2017-03-02:
Hello @gustavogini,

Try:

```c++
Serial.println(data);
JsonObject& root = jsonBuffer.parseObject(data);
```

It's crucial to show the content before calling `parseObject()` because the function will modify it.

bblanchon, 2017-03-09:
@gustavogini, any progress ?

closed 2017-03-15 by bblanchon

gustavogini, 2017-03-15:
Hello @bblanchon

Unfortunately no progress with json+nRF24L01. The nRF24L01 has the transmition limited in 32bytes becouse of this isn't possible to send the json.  Do you know some one that obtained success with this? Thanks!

bblanchon, 2017-03-15:
Did you try to send the JSON in chunks of 32 bytes?

gustavogini, 2017-03-15:
No. How can I do this? Does exist a FAQ about this?

Thanks!

gewisser, 2017-03-16:
Something like this...

```javascript
       const uint64_t pipe = 0xE8E8F0F0E1LL;
       const byte MAX_NRF_BUF = 32;
       const byte DATA_LEN = MAX_NRF_BUF - 2;
......

        int poco01 = 1;
	int poco02 = 2;
	int rele01 = 3;
	int rele02 = 4;

	StaticJsonBuffer<200> jsonBuffer;
	JsonObject& root = jsonBuffer.createObject();
	root["Poco01"] = poco01;
	root["Poco02"] = poco02;
	root["Rele01"] = rele01;
	root["Rele02"] = rele02;

	char buffer[200];
	int n = root.printTo(buffer, sizeof(buffer));

	byte c = n / DATA_LEN; // We learn how many packets in the message size DATA_LEN
	if (n % DATA_LEN != 0) c++; // We check whether there is a remainder, from division. If there is then increment by 1
	
	unsigned char nrf_pkg[MAX_NRF_BUF];

	for (size_t i = 1; i <= c; i++)
	{
		nrf_pkg[0] = c; // The first byte of the packet contains the total number of packets in the full message
		nrf_pkg[1] = i; // The second byte contains the sequence number of the packet
		
		byte offset = 0;
		if (i > 1)
			offset = DATA_LEN * (i - 1) + 1;

		// The remaining 30 (DATA_LEN) bytes are filled with the required data
		memcpy(&nrf_pkg[2], &buffer[offset], DATA_LEN);

		radio.write(nrf_pkg, MAX_NRF_BUF);
	}
```

gustavogini, 2017-03-16:
Hello @gewisser 

Thanks for your support! I will do the tests!

Regards!

gewisser, 2017-03-16:
It may be necessary to replace

```javascript
		if (i > 1)
			offset = DATA_LEN * (i - 1) + 1;
```
on

```javascript
		if (i > 1)
			offset = DATA_LEN * (i - 1);
```