bblanchon/ArduinoJson #457: Requesting URL: ParseObject() failed

mpcpro, 2017-03-02:
Hey.

Trying to get my Arduino Uno to parse json from openweathermap but it seems to fail. I am kinda new to arduino so I built my code from examples of others I've found. I am wondering what I am doing wrong, but looking at the code I am unable to see my error.  

I tried checking out the FAQ for parsing errors and none of them seem to work. I tried upping my StaticJsonBuffer, also tried switching from StaticJsonBuffer to DynamicJson, I've also tried switching the html type from 1.1 to 1.0. Also made sure that the json is Object and not Array (since the json starts on "{"). 

In my serial monitor I get this information back:

"
My IP Address is: 192.168.10.137
Connecting...
Connected!
connecting to api.openweathermap.org
Requesting URL: ParseObject() failed
0.00
"

I am using an Arduino Uno with an Ethernet Shield. 

My code is attached.

[openweathermaptest.txt](https://github.com/bblanchon/ArduinoJson/files/814997/openweathermaptest.txt)

bblanchon, 2017-03-03:
Hi @mpcpro,

I think it's impossible to parse openweathermap data with ArduinoJson on an Arduino Uno because it doesn't have enough RAM.

I tested with the [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/), and it says you need 1484 to parse the JSON, this is 72% of all your memory!

Moreover, your program uses much more memory.
The JSON input is present twice in memory: in `data` and `buffer`.
Those two variables take 2*915 = 1830 bytes, this is 89% of all your memory!!

To reduce the memory usage, let the `JsonBuffer` read directly from the `EthernetClient`, as in the [JsonHttpClient example](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonHttpClient/JsonHttpClient.ino):

```c++
StaticJsonBuffer<1500> jsonBuffer;
JsonObject& root = jsonBuffer.parseObject(client);
```

But this will work only work if there are at least 1500 bytes available in RAM.

Regards,
Benoit

mpcpro, 2017-03-03:
Thanks for your reply. I will give it a go, if that does not work I will try my ESP8266 ESP-12. Thanks a lot!

bblanchon, 2017-03-03:
You're welcome. Thanks for using ArduinoJson.

closed 2017-03-03 by bblanchon