bblanchon/ArduinoJson #459: I am facing problem regarding Arduinojson buffer reset or flush.

Joypacharya, 2017-03-03:
Hello 
I used both static and dynamicjson for data parses from MQTT.
After some time my json buffer overflow and my device did not response.
I actually did see the all your F.A.Q. in this link ,
https://bblanchon.github.io/ArduinoJson/faq/

Still i cant reset or flush my jsonbuffer.

In staticjson buffer,
const int sizeofbuffer = 1000;
StaticJsonBuffer<sizeofbuffer> jsonBuffer11; 
jsonBuffer11 = StaticJsonBuffer<sizeofbuffer>(); // reset the buffer as per defined into F.A.Q.

In dynamicjson buffer,

DynamicJsonBuffer jsonBuffer11;
JsonObject& root = jsonBuffer11.createObject();
jsonBuffer11 = 0; //reset the buffer (I am not sure about this one.)

I used this two terms simultaneously but still i do not sort out this issues.
Even check pitfall effect and other F.A.Q.
Please give me best solution for this one.

bblanchon, 2017-03-03:
Hello @Joypacharya,

Once again, there is no function to reset a `JsonBuffer` simply because it is a bad idea.
Don't reuse the same `JsonBuffer`.
Declare your `JsonBuffer` in a function scope, **do not make it global**.

See:

* [FAQ: What's the best way to use the library?](https://bblanchon.github.io/ArduinoJson/faq/whats-the-best-way-to-use-the-library/)
* [FAQ: Why shouldn't I use a global JsonBuffer?](https://bblanchon.github.io/ArduinoJson/faq/why-shouldnt-i-use-a-global-jsonbuffer/)
* [FAQ: How to reuse a JsonBuffer?](https://bblanchon.github.io/ArduinoJson/faq/how-to-reuse-a-jsonbuffer/)
* [Bag of Tricks: clear()](https://github.com/bblanchon/ArduinoJson/wiki/Bag-of-Tricks#reuse-jsonbuffer)

Joypacharya, 2017-03-05:
I tried all tricks still i am facing this issues.
So, check and reply me.

bblanchon, 2017-03-06:
Hello Joy,
Can you please copy/paste a minimalistic program that reproduces the problem?
Thanks.

closed 2017-03-15 by bblanchon