bblanchon/ArduinoJson #460: Compilation failure in platformio

dromer, 2017-03-03:
Version: 5.8.3
Target: esp8266
```
In file included from .piolibdeps/ArduinoJson_ID64/include/ArduinoJson.hpp:12:0,
from .piolibdeps/ArduinoJson_ID64/include/ArduinoJson.h:8,
from .piolibdeps/ArduinoJson_ID64/ArduinoJson.h:8,
from src/main.cpp:11:
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/JsonObject.hpp: In instantiation of 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::node_type* ArduinoJson::JsonObject::findNode(TStringRef) const [with TStringRef = const char&; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::node_type = ArduinoJson::In
ternals::ListNode<ArduinoJson::JsonPair>]':
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/JsonObject.hpp:300:47:   required from 'typename ArduinoJson::Internals::JsonVariantAs<TValue>::type ArduinoJson::JsonObject::get_impl(TStringRef) const [with TStringRef = const char&; TValue = ArduinoJson::JsonObject; typename ArduinoJson::Internals::JsonVariantAs
<TValue>::type = ArduinoJson::JsonObject&]'
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/JsonObject.hpp:168:48:   required from 'typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<TString>::value), typename ArduinoJson::Internals::JsonVariantAs<T>::type>::type ArduinoJson::JsonObject::get(const TString&) const [with TValue =
 ArduinoJson::JsonObject; TString = char; typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<TString>::value), typename ArduinoJson::Internals::JsonVariantAs<T>::type>::type = ArduinoJson::JsonObject&]'
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/JsonObjectSubscript.hpp:64:36:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = ArduinoJson::JsonObject; TStringRef = const char&; typename ArduinoJson::Internals::JsonVariant
As<T>::type = ArduinoJson::JsonObject&]'

.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Deserialization/../JsonVariantBase.hpp:63:35:   required from 'const typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = ArduinoJson::JsonObject; TImpl = ArduinoJson::JsonObjectSubscript<const char&>; ty
pename ArduinoJson::Internals::JsonVariantAs<T>::type = ArduinoJson::JsonObject&]'
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/Deserialization/../JsonVariantBase.hpp:116:14:   required from 'typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::Internals::StringTraits<const TString*>::has_equals, ArduinoJson::JsonObjectSubscript<const TString*> >::type ArduinoJson::JsonVariantBase<TImpl>
::operator[](const TString*) [with TString = char; TImpl = ArduinoJson::JsonObjectSubscript<const char&>; typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::Internals::StringTraits<const TString*>::has_equals, ArduinoJson::JsonObjectSubscript<const TString*> >::type = ArduinoJson::JsonObjectSubscript<const c
har*>]'
src/main.cpp:177:60:   required from here
.piolibdeps/ArduinoJson_ID64/include/ArduinoJson/JsonObject.hpp:291:77: error: 'equals' is not a member of 'ArduinoJson::Internals::StringTraits<const char&, void>'
if (Internals::StringTraits<TStringRef>::equals(key, node->content.key))
^
```

bblanchon, 2017-03-03:
Hi @dromer,

Can you please copy the content of line 177 of main.cpp?

Were you trying to read a `char`? [because it's not supported](https://bblanchon.github.io/ArduinoJson/faq/why-jsonvariant-cannot-be-converted-to-char/), you must choose `unsigned char` or `signed char`.

dromer, 2017-03-03:
Hey @bblanchon,

On this line I read a boolean from my json:
```
bool spaceState = root[mySpaces[thisSpace]]["state"]["open"];
```
'root' here is the parsed jsonBuffer object, 'mySpaces' is an array of strings. and 'thisSpace' is an integer iteration over this array.

Btw this code worked fine when I had everything in a single main.cpp, but now I moved some functions to separate files and headers.

bblanchon, 2017-03-03:
I reproduced the error with the following:

```c++
bool spaceState = root['a'];
```

This error is logical since the key of a `JsonObject` is supposed to be a string, not a char.

In your program, I think `mySpaces` is not an array of string, but an array of char.

dromer, 2017-03-04:
Ah sorry, you are right.
My array is defined as such:
```
  const char *mySpaces[] = {
      "test1",
      "test2",
      "test3"
  };
```
Here an example that works: https://gist.github.com/dromer/c7121e4c273bdc8b1715cc8059a0ffa1

bblanchon, 2017-03-05:
@dromer, can I close this issue?

dromer, 2017-03-05:
Yeah seems I've got the declarations in order and can use the method I wanted.
Thnx for the help.

closed 2017-03-05 by dromer

bblanchon, 2017-03-06:
You're welcome @dromer 😄 
Thank you for using ArduinoJson.