bblanchon/ArduinoJson #462: Affect 0 integer crash on ESP8266

hallard, 2017-03-05:
I'm using latest git of arduino-esp8266 and ArduinoJson and when I'm creating JSON sometime when ESP8266 is crashing, I spent some time too identify the cause but now I'm able to reproduce the problem

Here one of my part of my configuration structure, look at it it's important for type declared
```cpp
// Config for emoncms
// 128 Bytes
typedef struct 
{
  char  host[CFG_EMON_HOST_SIZE+1]; 	
  char  apikey[CFG_EMON_APIKEY_SIZE+1]; 
  char  url[CFG_EMON_URL_SIZE+1];  		
  uint16_t port;   						
  uint16_t node;     						
  uint32_t freq;  
  uint8_t filler[21];
} _emoncms;
```

Here the code having problem

```cpp
  AsyncJsonResponse * response = new AsyncJsonResponse();
  JsonObject& root = response->getRoot(); 

  root[FPSTR(CFG_SSID)]    = config.ssid;
  root[FPSTR(CFG_PSK)]     = config.psk;
  root[FPSTR(CFG_HOST)]    = config.host ;
  root[FPSTR(CFG_AP_PSK)]  = config.ap_psk ;
  root[FPSTR(CFG_AP_SSID)] = config.ap_ssid ;

  ip_addr=config.ip;   root[FPSTR(CFG_IP)]      = ip_addr.toString();
  ip_addr=config.mask; root[FPSTR(CFG_MASK)]    = ip_addr.toString();
  ip_addr=config.gw;   root[FPSTR(CFG_GATEWAY)] = ip_addr.toString();
  ip_addr=config.dns;  root[FPSTR(CFG_DNS)]     = ip_addr.toString();
  root[FPSTR(CFG_EMON_HOST)] = config.emoncms.host;
  root[FPSTR(CFG_EMON_PORT)] = config.emoncms.port;
  root[FPSTR(CFG_EMON_URL)]  = config.emoncms.url;
  root[FPSTR(CFG_EMON_KEY)]  = config.emoncms.apikey;
  root[FPSTR(CFG_EMON_NODE)] = config.emoncms.node;
  root[FPSTR(CFG_EMON_FREQ)] =  config.emoncms.freq;
```
The last 2 lines make the ESP8266 to crash (reset) but to avoid this, I'm casting the 2 last lines as int.
Now it works, I thought it was a cast problem, but what is strange is that the line `config.emoncms.port` which is also `uint16_t` but not casted works ?!?

So the following code works
```
  root[FPSTR(CFG_EMON_NODE)] = (unsigned int) config.emoncms.node;
  root[FPSTR(CFG_EMON_FREQ)] =  (unsigned int) config.emoncms.freq;
```

Any idea ?

bblanchon, 2017-03-06:
Hello Charles,

It looks to me that you're running out of memory.
Are you sure that `response` is deleted?
Why don't you use static allocation for `response`?
Can you check the size of the `JsonBuffer`?

See:

* [FAQ: Why does my device crash or reboot?](https://bblanchon.github.io/ArduinoJson/faq/why-does-my-device-crash-or-reboot/)
* [API Reference: JsonBuffer::size()](https://bblanchon.github.io/ArduinoJson/api/jsonbuffer/size/)

hallard, 2017-03-06:
Benoit,
Yes I thought this at first time but Memory looks good, here the debug output showing free ram on each stage when I cast each integer data 

```
Serving /config page...
Stage 1 : 21520
Stage 2 : 20992
Stage 3 : 20992
Stage 4 : 19952
Stage 5 : 19952
Stage 6 : 19952
Stage 7 : 19952
Stage 8 : 19952
```
So as you can see all is working fine with casting.

the full code is
```cpp
void confJSONTable(AsyncWebServerRequest * request)
{
  IPAddress ip_addr;
  size_t l;
  String str;

  AsyncJsonResponse * response = new AsyncJsonResponse();
  JsonObject& root = response->getRoot(); 

  DebuglnF("Serving /config page...");

  root[FPSTR(CFG_SSID)]    = config.ssid;
  root[FPSTR(CFG_PSK)]     = config.psk;
  root[FPSTR(CFG_HOST)]    = config.host ;
  root[FPSTR(CFG_AP_PSK)]  = config.ap_psk ;
  root[FPSTR(CFG_AP_SSID)] = config.ap_ssid ;

  DebugF("Stage 1 : ")
  Debugln(ESP.getFreeHeap());

  ip_addr=config.ip;   root[FPSTR(CFG_IP)]      = ip_addr.toString();
  ip_addr=config.mask; root[FPSTR(CFG_MASK)]    = ip_addr.toString();
  ip_addr=config.gw;   root[FPSTR(CFG_GATEWAY)] = ip_addr.toString();
  ip_addr=config.dns;  root[FPSTR(CFG_DNS)]     = ip_addr.toString();
  root[FPSTR(CFG_EMON_HOST)] = config.emoncms.host;
  root[FPSTR(CFG_EMON_PORT)] = (unsigned int) config.emoncms.port;
  root[FPSTR(CFG_EMON_URL)]  = config.emoncms.url;
  root[FPSTR(CFG_EMON_KEY)]  = config.emoncms.apikey;
  root[FPSTR(CFG_EMON_NODE)] = (unsigned int) config.emoncms.node;
  root[FPSTR(CFG_EMON_FREQ)] = (unsigned int) config.emoncms.freq;

  DebugF("Stage 2 : ")
  Debugln(ESP.getFreeHeap());

  root[FPSTR(CFG_JDOM_HOST)] = config.jeedom.host;
  root[FPSTR(CFG_JDOM_PORT)] = (unsigned int) config.jeedom.port;
  root[FPSTR(CFG_JDOM_URL)]  = config.jeedom.url;
  root[FPSTR(CFG_JDOM_KEY)]  = config.jeedom.apikey;
  root[FPSTR(CFG_JDOM_ADCO)] = config.jeedom.adco;
  root[FPSTR(CFG_JDOM_FREQ)] = (unsigned int) config.jeedom.freq;

  DebugF("Stage 3 : ")
  Debugln(ESP.getFreeHeap());
  root[FPSTR(CFG_DOMZ_HOST)] = config.domz.host;
  root[FPSTR(CFG_DOMZ_PORT)] = (unsigned int) config.domz.port;
  root[FPSTR(CFG_DOMZ_URL)]  = config.domz.url;
  root[FPSTR(CFG_DOMZ_USER)] = config.domz.user;
  root[FPSTR(CFG_DOMZ_PASS)] = config.domz.pass;
  root[FPSTR(CFG_DOMZ_INDEX)]= (unsigned int) config.domz.index;
  root[FPSTR(CFG_DOMZ_FREQ)] = (unsigned int) config.domz.freq;

  DebugF("Stage 4 : ")
  Debugln(ESP.getFreeHeap());

  root[FPSTR(CFG_COUNT1_GPIO)]  = (unsigned int) config.counter.gpio1;
  root[FPSTR(CFG_COUNT1_DELAY)] = (unsigned int) config.counter.delay1;
  root[FPSTR(CFG_COUNT1_VALUE)] = (unsigned long) config.counter.value1;
  root[FPSTR(CFG_COUNT2_GPIO)]  = (unsigned int) config.counter.gpio2;
  root[FPSTR(CFG_COUNT2_DELAY)] = (unsigned int) config.counter.delay2;
  root[FPSTR(CFG_COUNT2_VALUE)] = (unsigned long) config.counter.value2;

  DebugF("Stage 5 : ")
  Debugln(ESP.getFreeHeap());

  root[FPSTR(CFG_OTA_AUTH)] = config.ota_auth;
  root[FPSTR(CFG_OTA_PORT)] = (unsigned int) config.ota_port;
  root[FPSTR(CFG_SENS_SI7021)]= (unsigned int) config.sensors.en_si7021;
  root[FPSTR(CFG_SENS_SHT10)] = (unsigned int) config.sensors.en_sht10;
  root[FPSTR(CFG_SENS_MCP3421)] = (unsigned int) config.sensors.en_mcp3421;

  DebugF("Stage 6 : ")
  Debugln(ESP.getFreeHeap());

  root[FPSTR(CFG_CFG_RGBLED)] = (config.config&CFG_RGB_LED)?1:0;
  root[FPSTR(CFG_CFG_DEBUG)]  = (config.config&CFG_DEBUG)?1:0;;
  root[FPSTR(CFG_CFG_OLED)]   = (config.config&CFG_LCD)?1:0;;
  root[FPSTR(CFG_CFG_AP)]     = (config.config&CFG_AP)?1:0;;
  root[FPSTR(CFG_CFG_WIFI)]   = (config.config&CFG_WIFI)?1:0;;
  root[FPSTR(CFG_CFG_STATIC)] = (config.config&CFG_STATIC)?1:0;;

  DebugF("Stage 7 : ")
  Debugln(ESP.getFreeHeap());

  root[FPSTR(CFG_LED_BRIGHTNESS)] = (unsigned int) config.led_bright;
  root[FPSTR(CFG_LED_HEARTBEAT)]  = (unsigned int) config.led_hb;
  root[FPSTR(CFG_LED_NUM)]        = (unsigned int) config.led_num;
  root[FPSTR(CFG_LED_GPIO)]       = (unsigned int) config.led_gpio;
  root[FPSTR(CFG_LED_TYPE)]       = (unsigned int) config.led_type;

  DebugF("Stage 8 : ")
  Debugln(ESP.getFreeHeap());

  root[FPSTR(CFG_SENS_FREQ)] = (unsigned int) config.sensors.freq;
  str = config.sensors.hum_min_warn; str += "," ; str += config.sensors.hum_max_warn;
  root[FPSTR(CFG_SENS_HUM_LED)]   = str;
  str = config.sensors.temp_min_warn; str += "," ; str += config.sensors.temp_max_warn;
  root[FPSTR(CFG_SENS_TEMP_LED)]  = str;
  str = config.sensors.pwr_min_warn; str += "," ; str += config.sensors.pwr_max_warn;
  root[FPSTR(CFG_SENS_PWR_LED)]  = str;

  size_t jsonlen ;
  jsonlen = response->setLength();
  request->send(response);
  //Debugf("Json size %lu bytes\r\n", jsonlen);
}
```

Now if I just change the lines in stage 1

```cpp
  root[FPSTR(CFG_EMON_NODE)] = (unsigned int) config.emoncms.node;
  root[FPSTR(CFG_EMON_FREQ)] = (unsigned int) config.emoncms.freq;
```
by (only removed cast)

```cpp
  root[FPSTR(CFG_EMON_NODE)] = config.emoncms.node;
  root[FPSTR(CFG_EMON_FREQ)] = config.emoncms.freq;
```

Here the output 
```
Serving /config page...
Stage 1 : 21496

Exception (9):
epc1=0x4020de60 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff0d6d depc=0x00000000

ctx: sys
sp: 3ffffbb0 end: 3fffffb0 offset: 01a0
```

The declaration of the buffer seems to be done on AsyncWebServer library written by @me-no-dev and I'm using the advanced one as the documentation says here 
https://github.com/me-no-dev/ESPAsyncWebServer#arduinojson-basic-response

hallard, 2017-03-06:
Just hooked @me-no-dev library to get the JSON buffer size so I can display it
when casted and all working
```
Serving /config page...
Stage 1 : JSONBufferSize:123  FreeRam:21168
Stage 2 : JSONBufferSize:406  FreeRam:20640
Stage 3 : JSONBufferSize:574  FreeRam:20640
Stage 4 : JSONBufferSize:770  FreeRam:19600
Stage 5 : JSONBufferSize:938  FreeRam:19600
Stage 6 : JSONBufferSize:1081  FreeRam:19600
Stage 7 : JSONBufferSize:1243  FreeRam:19600
Stage 8 : JSONBufferSize:1393  FreeRam:19600
```

when removed some cast in stage 1
```
Serving /config page...
Stage 1 : JSONBufferSize:123  FreeRam:21168

Exception (9):
epc1=0x4020de5c epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff0ecd depc=0x00000000

ctx: sys
sp: 3ffffbb0 end: 3fffffb0 offset: 01a0
```

bblanchon, 2017-03-06:
This definitely looks like a bug.

What is the type of value that triggers the bug `short`, `int`, `long`, `long long`?

Can you locate the line of the crash?
Is it when you set the signed value?
Or is it when `printTo()` is called?
(BTW, you should use the excellent [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) of @me-no-dev)

Can you reproduce the bug with a smaller program?
Ideally, just a few lines not involving ESPAsyncWebServer.
You can start from [JsonGeneratorExample.ino](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonGeneratorExample/JsonGeneratorExample.ino).

hallard, 2017-03-06:
the line (in my code) 
```
root[FPSTR(CFG_EMON_NODE)] = config.emoncms.node;
```
Value is an `uint16_t`, I'll try to investigate more ASAP, but as casting solved my issue I'm focusing on my main code for now.

Stack trace is interesting, thanks for the tips, seems to be a structure alignment, I already add this kind of problem. Since uint16_t is 2 bytes somewhere in code we need to be aligned on 4 bytes boundaries (for ESP8266), casting as int seems do the trick.

```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
Decoding 30 results
0x4020ec20: JsonVariant  at S:\Skydrive\devt\Arduino\libraries\ArduinoJson/include/ArduinoJson/Deserialization/../JsonVariant.hpp line 98
:  (inlined by) set  at S:\Skydrive\devt\Arduino\libraries\ArduinoJson/include/ArduinoJson/Data/ValueSetter.hpp line 23
:  (inlined by) set_impl  at S:\Skydrive\devt\Arduino\libraries\ArduinoJson/include/ArduinoJson/JsonObject.hpp line 317
:  (inlined by) ArduinoJson::TypeTraits::EnableIf::value, bool>::type ArduinoJson::JsonObject::set (__FlashStringHelper const*, unsigned short const&) at S:\Skydrive\devt\Arduino\libraries\ArduinoJson/include/ArduinoJson/JsonObject.hpp line 122
0x40254c54: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4020ec1d: set_impl  at S:\Skydrive\devt\Arduino\libraries\ArduinoJson/include/ArduinoJson/JsonObject.hpp line 314
:  (inlined by) ArduinoJson::TypeTraits::EnableIf::value, bool>::type ArduinoJson::JsonObject::set (__FlashStringHelper const*, unsigned short const&) at S:\Skydrive\devt\Arduino\libraries\ArduinoJson/include/ArduinoJson/JsonObject.hpp line 122
0x4020ef43: operator=  at S:\Skydrive\devt\Arduino\libraries\ArduinoJson/include/ArduinoJson/JsonObjectSubscript.hpp line 54
:  (inlined by) confJSONTable(AsyncWebServerRequest*) at C:\Users\CHARLE~1\AppData\Local\Temp\arduino_build_974766\sketch/webserver.cpp line 500
0x40216edc: AsyncWebServerRequest::_parseReqHeader() at S:\Skydrive\devt\Arduino\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 740
0x4021f1fc: std::_Function_handler ::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*) at s:\skydrive\devt\arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x4021a010: std::function ::operator()(AsyncWebServerRequest*) const at S:\Skydrive\devt\Arduino\libraries\ESPAsyncWebServer\src/WebServer.cpp line 79
:  (inlined by) AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at S:\Skydrive\devt\Arduino\libraries\ESPAsyncWebServer\src/WebHandlerImpl.h line 91
0x4021d24f: String::reserve(unsigned int) at S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 720
0x4021d59b: String::concat(char const*, unsigned int) at S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 720
0x40218068: AsyncWebServerRequest::_parseLine() at S:\Skydrive\devt\Arduino\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 740
0x4021d5f8: String::concat(char const*) at S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266\cores\esp8266/WString.cpp line 720
0x40218428: AsyncWebServerRequest::_onData(void*, unsigned int) at S:\Skydrive\devt\Arduino\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 740
0x4010068c: free at S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x401068f8: vPortFree at S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266\cores\esp8266/heap.c line 47
0x4010020c: _umm_free at S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x40218471: _M_invoke at S:\Skydrive\devt\Arduino\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 740
0x40246e7d: system_pp_recycle_rx_pkt at ?? line ?
0x40213d94: AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at S:\Skydrive\devt\Arduino\libraries\ESPAsyncTCP\src/ESPAsyncTCP.cpp line 650
0x40228eb1: pbuf_free at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/pbuf.c line 909 (discriminator 3)
0x40213de7: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at S:\Skydrive\devt\Arduino\libraries\ESPAsyncTCP\src/ESPAsyncTCP.cpp line 650
0x4022e36c: tcp_input at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/tcp_in.c line 394 (discriminator 1)
0x401068e8: pvPortMalloc at S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266\cores\esp8266/heap.c line 40
0x40220000: std::function ::swap(std::function &) at ?? line ?
0x4022ca31: ip_input at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/ipv4/ip.c line 559
0x4022be01: ethernet_input at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/netif/etharp.c line 1379
0x40231b57: pp_tx_idle_timeout at ?? line ?
0x40231aca: pp_tx_idle_timeout at ?? line ?
0x4024681b: ets_snprintf at ?? line ?

```

bblanchon, 2017-03-06:
That's all I need to fix this bug.
Thanks Charles 👍

bblanchon, 2017-03-06:
Hi Charles,

I've not been able to reproduce the bug 😞 

The only way I could get something similar is by setting `ARDUINOJSON_ENABLE_ALIGNMENT` to `0`.
Can you verify that this constant is set to `1`?
Can you provide a complete program to reproduce the bug?
Can you check the version of ArduinoJson?
Can you check the version of the Arduino Core for ESP8266? (I'm using 2.3.0)

hallard, 2017-03-07:
Benoit,
I found the issue, it's a structure alignement, I got old code that was sending structure content over the air with RF modules, and to be compatible between the sender (Arduino) and receiver (ESP8266) the structure needed to be aligned on 1 byte both sides (to get same data lenght and organization). To do this I was using pack pragma pack directive as follow on ESP8266

```cpp
#pragma pack(push)  // push current alignment to stack
#pragma pack(1)     // set alignment to 1 byte boundary

typedef struct 
{
  char  host[CFG_EMON_HOST_SIZE+1]; 		// FQDN 
  char  apikey[CFG_EMON_APIKEY_SIZE+1]; // Secret
  char  url[CFG_EMON_URL_SIZE+1];  			// Post URL
  uint16_t port;   									    // Protocol port (HTTP/HTTPS)
  uint16_t node;     									  // optional node
  uint32_t freq;                        // refresh rate
  uint8_t filler[21];									  // in case adding data in config avoiding loosing current conf by bad crc*/
} _emoncms;

#pragma pack(pop)     // set alignment to default
```

I removed this pragma directive from my code and now it's working.
But I don't like it, just because it's increasing structure size due to ESP8266 default alignment and my structure is huge so I don't want to.

I'll try to give you an example but trying something like this should reproduce the bug

```cpp
#pragma pack(push)  // push current alignment to stack
#pragma pack(1)     // set alignment to 1 byte boundary

// Config for emoncms
// 128 Bytes
typedef struct 
{
  char  host[CFG_EMON_HOST_SIZE+1]; 		// FQDN 
  char  apikey[CFG_EMON_APIKEY_SIZE+1]; // Secret
  char  url[CFG_EMON_URL_SIZE+1];  			// Post URL
  uint32_t port;   									    // Protocol port (HTTP/HTTPS)
  uint32_t node;     									  // optional node
  uint32_t freq;                        // refresh rate
  uint8_t filler[17];									  // in case adding data in config avoiding loosing current conf by bad crc*/
} _emoncms;
#pragma pack(pop) 

// Config saved into eeprom
// 1024 bytes total including CRC
typedef struct 
{
  _emoncms emoncms;        // Emoncms configuration
  uint8_t  filler1[894];          // Another filler in case we need more
  uint16_t crc;
} _Config;

_Config config;

root["host"] = config.emoncms.host;
root["port"] = config.emoncms.port;
root["url"]  = config.emoncms.url;
root["key"]  = config.emoncms.apikey;
root["node"] = config.emoncms.node;
root["freq"] = config.emoncms.freq;
```

I'm using latest git (so master not release) of ArduinoJson and Arduino-ESP8266 and yes `ARDUINOJSON_ENABLE_ALIGNMENT ` is set to 1

bblanchon, 2017-03-07:
Well, that makes sense:

1. The `LoadStoreAlignmentCause` was triggered by the reading of the unaligned integer.
2. `JsonVariant` was pointed by the stack trace because the argument is passed by reference
3. Casting to `unsigned int` muted the bug because a temporary copy was made on the stack.

Charles, I think I can close this bug.
What do you think?

hallard, 2017-03-07:
Benoit,
Yes, no pb I closed it

closed 2017-03-07 by hallard