bblanchon/ArduinoJson #463: Deleting JSON buffers and objects before declaring new ones

MCUdude, 2017-03-06:
Hi!
As I've understood, there's no way of reusing a JSON buffer.

I'm creating a project on an Arduino where small JSON messages are sent over RS485 data lines. If a spesific command is received (`dump_data`), all the content should be sent back. To prevent the MCU from going out of ram, each message is no more that 100 byte large. I want to create a functin that sends all the content divided into small JSON payloads. In total there might be upto 15 different messages.
Here's an exaple of a `sendAll()` function

``` c++
void sendAll()
{
  // Create JSON buffer
  StaticJsonBuffer<100> jsonBuffer;
  
  // Create object
  JsonObject& root = jsonBuffer.createObject();
  
  
  // Send volume
  root[F("cmd")] = F("volume");
  root[F("volume")] = dsp.volumeData;
  root[F("slew")] = dsp.volumeSlewData;
  rs485.send(jsonAck, jsonBuffer);

 // Send mux
 // Clear or delete JSON buffer here
  root[F("cmd")] = F("mux");
  root[F("muxdata")] = dsp.muxdata;
  rs485.send(jsonAck, jsonBuffer);

//..
}
```

In total there are 15 messages like this that's going to be sent over RS485 in the same function. I hope I don't have to either create a new buffer for each message without getting rid of the old one. this will allocate so much RAM, and the MCU will crash. How can this be solved? Can the JSON buffer and the root object be explicy deleted before creating a new buffer and object?

MCUdude, 2017-03-07:
Well, this is embarrasing, why didn't I even think of that!

Thanks 😁

closed 2017-03-07 by MCUdude

bblanchon, 2017-03-07:
You're welcome 😄 
Thank you for using ArduinoJson.