bblanchon/ArduinoJson #465: Type change to long long

b17460904, 2017-03-07:
I want to using unix millisecond timestamp as Jason vale,but it seems like the type is long, it will overflow, how to change long type to long long type? Thanks.

bblanchon, 2017-03-07:
Hello @b17460904,

You can define [ARDUINOJSON_USE_LONG_LONG](https://bblanchon.github.io/ArduinoJson/api/config/use_long_long/) to `1` to switch to `long long`.

b17460904, 2017-03-08:
Hi, bblanchon,

Thank for your help. After define  ARDUINOJSON_USE_LONG_LONG to 1, the program have an error " hpp:50:30: error: 'strtoull' was not declared in this scope return strtoull(s, NULL, 10);".
Are there any criteria for using long long format on Arduino Mega?

bblanchon, 2017-03-08:
Hi @b17460904,

This means that the Arduino Core for AVR has not been updated for `long long`.

I plan on dropping `strtoul()` in the next revision anyway.
Until then, I'm afraid I don't have any workaround 😞

b17460904, 2017-03-09:
Hi @bblanchon ,

OK, I Know. Thanks for your kindly response.

closed 2017-03-25 by bblanchon