bblanchon/ArduinoJson #466: JSON & C++ Object

StephaneRod, 2017-03-09:
I am very new in JSON and I am working on a DIY project on a DUE and so many objects that I need to dynamically allocate them only when needed one by one. 

Is there any example you could give us where we can understand how to use your library to serialize and deserialize C++ objects and their relations (pointers, serialiaze graph of object)).

I only see serialization of some "Struct" but is there a kind of parser that could read and serialize like the JSONExport of ThorsSerializer ?

example 

class person
age
dad
mam

person jacques(52, John, Kelly)

jsonexport(jacques); ?

Thank you 
stephane

bblanchon, 2017-03-09:
Hello Stephane,
What did you try?

StephaneRod, 2017-03-10:
Hi Benoit,

thanks for your kind email.

I tried your examples

but I see that I have to build functions myself that will save/load each type of objects

and maybe because of my misunderstanding of your functions there is something else

regards

stephane

bblanchon, 2017-03-10:
Hello Stephane,

If I understand correctly, you're looking for an automated way to transform a `struct` to JSON and vice-versa, like you'd do in C# or Java.

Unfortunately, due to the lack of introspection in C++, it's impossible to do that natively.
The C++ community has produced a few libraries (like [Boost.Fusion](http://www.boost.org/doc/libs/1_61_0/libs/fusion/doc/html/fusion/adapted/define_struct.html), [CAMP](https://github.com/tegesoft/camp) and [Ponder](https://github.com/billyquith/ponder)) that provide a reflection-like feature.

ArduinoJson doesn't provide such feature, so you need to do this part by manually.

Regards,
Benoit

StephaneRod, 2017-03-10:
Hello Benoit,
Thanks for your feed back
I will make it manually with your library
Thank you
Stéphane

bblanchon, 2017-03-10:
You're welcome Stéphane. 
Thank you for using ArduinoJson.

closed 2017-03-10 by bblanchon