bblanchon/ArduinoJson #467: no match for 'operator[]'

awaing-wmsu, 2017-03-10:
Hi, I seems to encounter this kind of error. 

> no match for 'operator[]' (operand types are 'ArduinoJson::JsonObject' and 'String')
[image]

`Arduino: 1.8.0 (Windows 10), Board: "Intel® Edison"

C:\Users\christian\Documents\Arduino\Test\Test.ino: In function 'String get_gpio_status()':

Test:221: error: no match for 'operator[]' (operand types are 'ArduinoJson::JsonObject' and 'String')

   data[String(GPIO1_PIN)] = gpioState[0] ? true : false;

       ^

C:\Users\christian\Documents\Arduino\Test\Test.ino:221:7: note: candidates are:

In file included from C:\Users\christian\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson.h:9:0,

                 from C:\Users\christian\Documents\Arduino\libraries\ArduinoJson/ArduinoJson.h:7,

                 from C:\Users\christian\Documents\Arduino\Test\Test.ino:1:

C:\Users\christian\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonObject.hpp:51:16: note: ArduinoJson::JsonVariant& ArduinoJson::JsonObject::operator[](ArduinoJson::JsonObject::key_type)

   JsonVariant &operator[](key_type key);

                ^

C:\Users\christian\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonObject.hpp:51:16: note:   no known conversion for argument 1 from 'String' to 'ArduinoJson::JsonObject::key_type {aka const char*}'

C:\Users\christian\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonObject.hpp:55:22: note: const ArduinoJson::JsonVariant& ArduinoJson::JsonObject::operator[](ArduinoJson::JsonObject::key_type) const

   const JsonVariant &operator[](key_type key) const { return at(key); }

                      ^

C:\Users\christian\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonObject.hpp:55:22: note:   no known conversion for argument 1 from 'String' to 'ArduinoJson::JsonObject::key_type {aka const char*}'

Test:222: error: no match for 'operator[]' (operand types are 'ArduinoJson::JsonObject' and 'String')

   data[String(GPIO2_PIN)] = gpioState[1] ? true : false;

       ^

C:\Users\christian\Documents\Arduino\Test\Test.ino:222:7: note: candidates are:

In file included from C:\Users\christian\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson.h:9:0,

                 from C:\Users\christian\Documents\Arduino\libraries\ArduinoJson/ArduinoJson.h:7,

                 from C:\Users\christian\Documents\Arduino\Test\Test.ino:1:

C:\Users\christian\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonObject.hpp:51:16: note: ArduinoJson::JsonVariant& ArduinoJson::JsonObject::operator[](ArduinoJson::JsonObject::key_type)

   JsonVariant &operator[](key_type key);

                ^

C:\Users\christian\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonObject.hpp:51:16: note:   no known conversion for argument 1 from 'String' to 'ArduinoJson::JsonObject::key_type {aka const char*}'

C:\Users\christian\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonObject.hpp:55:22: note: const ArduinoJson::JsonVariant& ArduinoJson::JsonObject::operator[](ArduinoJson::JsonObject::key_type) const

   const JsonVariant &operator[](key_type key) const { return at(key); }

                      ^

C:\Users\christian\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonObject.hpp:55:22: note:   no known conversion for argument 1 from 'String' to 'ArduinoJson::JsonObject::key_type {aka const char*}'

Multiple libraries were found for "WiFiClient.h"
 Used: C:\Users\christian\AppData\Local\Arduino15\packages\Intel\hardware\i686\1.6.7+1.0\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
no match for 'operator[]' (operand types are 'ArduinoJson::JsonObject' and 'String')

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

I have followed the codes here: [Here](https://blog.thingsboard.io/2017/01/esp8266-gpio-control-over-mqtt-using.html)

and it works for them. But when I tried to use their code as the basis of my code, I am encountering this kind of issue. I am using Intel Edison. At first I encounter the c_str() error and I fix it by adding a code in the WString.h 

I hope you could help me fix this.  I am using the latest ArduinoJSON library.  Thank you.

bblanchon, 2017-03-10:
Hello Alexander,

I tried to compile the [StringExample.ino](https://github.com/bblanchon/ArduinoJson/blob/master/examples/StringExample/StringExample.ino) with Arduino IDE 1.8.1 and [Edison core 1.6.9+1.0](https://github.com/01org/corelibs-edison).
It compiles without error.

ArduinoJson enables the support for `String` via [`ARDUINOJSON_ENABLE_ARDUINO_STRING`](https://bblanchon.github.io/ArduinoJson/api/config/enable_arduino_string/).
Can you check that it is properly defined to `1`?

awaing-wmsu, 2017-03-10:
Hi, where can I find that ARDUINOJSON_ENABLE_ARDUINO_STRING if it is defined to 1?

awaing-wmsu, 2017-03-10:
Hi, I'm back. I have already installed IDE 1.8.1 and Edison Core 1.6.9+1.0 
then tried compiling StringExample.ino and it compiles with errors :(

[image]

```
Arduino: 1.8.1 (Windows 10), Board: "Intel® Edison"

C:\Users\ITSMEA~1\AppData\Local\Temp\arduino_modified_sketch_994136\sketch_mar10a.ino: In function 'void setup()':

sketch_mar10a:24: error: no matching function for call to 'ArduinoJson::DynamicJsonBuffer::parseObject(String&)'

   JsonObject& root = jsonBuffer.parseObject(input);

                                                  ^

C:\Users\ITSMEA~1\AppData\Local\Temp\arduino_modified_sketch_994136\sketch_mar10a.ino:24:50: note: candidate is:

In file included from C:\Users\itsmealex\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/DynamicJsonBuffer.hpp:9:0,

                 from C:\Users\itsmealex\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson.h:7,

                 from C:\Users\itsmealex\Documents\Arduino\libraries\ArduinoJson/ArduinoJson.h:7,

                 from C:\Users\ITSMEA~1\AppData\Local\Temp\arduino_modified_sketch_994136\sketch_mar10a.ino:8:

C:\Users\itsmealex\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonBuffer.hpp:62:15: note: ArduinoJson::JsonObject& ArduinoJson::JsonBuffer::parseObject(char*, uint8_t)

   JsonObject &parseObject(char *json, uint8_t nestingLimit = DEFAULT_LIMIT);

               ^

C:\Users\itsmealex\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonBuffer.hpp:62:15: note:   no known conversion for argument 1 from 'String' to 'char*'

sketch_mar10a:28: error: no match for 'operator[]' (operand types are 'ArduinoJson::JsonObject' and 'String')

   long time = root[String("time")];

                   ^

C:\Users\ITSMEA~1\AppData\Local\Temp\arduino_modified_sketch_994136\sketch_mar10a.ino:28:19: note: candidates are:

In file included from C:\Users\itsmealex\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson.h:9:0,

                 from C:\Users\itsmealex\Documents\Arduino\libraries\ArduinoJson/ArduinoJson.h:7,

                 from C:\Users\ITSMEA~1\AppData\Local\Temp\arduino_modified_sketch_994136\sketch_mar10a.ino:8:

C:\Users\itsmealex\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonObject.hpp:51:16: note: ArduinoJson::JsonVariant& ArduinoJson::JsonObject::operator[](ArduinoJson::JsonObject::key_type)

   JsonVariant &operator[](key_type key);

                ^

C:\Users\itsmealex\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonObject.hpp:51:16: note:   no known conversion for argument 1 from 'String' to 'ArduinoJson::JsonObject::key_type {aka const char*}'

C:\Users\itsmealex\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonObject.hpp:55:22: note: const ArduinoJson::JsonVariant& ArduinoJson::JsonObject::operator[](ArduinoJson::JsonObject::key_type) const

   const JsonVariant &operator[](key_type key) const { return at(key); }

                      ^

C:\Users\itsmealex\Documents\Arduino\libraries\ArduinoJson/include/../include/ArduinoJson/JsonObject.hpp:55:22: note:   no known conversion for argument 1 from 'String' to 'ArduinoJson::JsonObject::key_type {aka const char*}'

sketch_mar10a:32: error: no match for 'operator[]' (operand types are 'ArduinoJson::JsonObject' and 'String')

   root[String("time")] = time;

       ^

[... 162 lines cut ...]
C:\Users\itsmealex\AppData\Local\Arduino15\packages\Intel\hardware\i686\1.6.7+1.0\cores\arduino/WString.h:66:11: note: String::String(unsigned char, unsigned char)

  explicit String(unsigned char, unsigned char base=10);

           ^

C:\Users\itsmealex\AppData\Local\Arduino15\packages\Intel\hardware\i686\1.6.7+1.0\cores\arduino/WString.h:65:11: note: String::String(char)

  explicit String(char c);

           ^

C:\Users\itsmealex\AppData\Local\Arduino15\packages\Intel\hardware\i686\1.6.7+1.0\cores\arduino/WString.h:59:2: note: String::String(const char*)

  String(const char *cstr = "");

  ^

exit status 1
no matching function for call to 'ArduinoJson::DynamicJsonBuffer::parseObject(String&)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

bblanchon, 2017-03-10:
To test the value, compile:

```c++
#include <ArduinoJson.h>

#if ARDUINOJSON_ENABLE_ARDUINO_STRING
#error ARDUINOJSON_ENABLE_ARDUINO_STRING is set to 1, which is OK
#else
#error  ARDUINOJSON_ENABLE_ARDUINO_STRING is set to 0, which is wrong.
#endif
```

to force to 1, do:

```c++
#define ARDUINOJSON_ENABLE_ARDUINO_STRING 1
#include <ArduinoJson.h>
```

bblanchon, 2017-03-15:
@awaing-wmsu Did it work ?

bblanchon, 2017-03-21:
@awaing-wmsu Did you get it working?

closed 2017-03-27 by bblanchon

itsmealex56, 2018-02-11:
Sorry, it took longer for me to reply. Yes, Bblanchon. Thank for your reply :)