bblanchon/ArduinoJson #468: What is the best war parse json string like below

SanjayGautam, 2017-03-21:
Hi Blanchon , 

Thanks for sharing the code it is very helpful , I am trying to parse the string like below from the rest service 

{
  "mac" : null,
  "lcId" : 634,
  "ln" : "VFF01",
  "ld" : "This is fun",
  "at" : 23,
  "ct" : 23,
  "lnBrkIntvls" : [ {
    "lnbrkIntrvlId" : 637,
    "sftNm" : "shift2",
    "sftStrtHhMn" : "15:10",
    "sftEndHhMn" : "11:00",
    "brkFrmHhMn" : "19:00",
    "brkToHhMn" : "19:15"
  }, {
    "lnbrkIntrvlId" : 638,
    "sftNm" : "shift1",
    "sftStrtHhMn" : "15:10",
    "sftEndHhMn" : "11:00",
    "brkFrmHhMn" : "20:00",
    "brkToHhMn" : "20:15"
  } ]
}

if you see that sometime the inBrkIntrvls are going to be 2 or 3 or 4 based on the configuration , I am more of the java guy and hence dotn know how I can make  dynamically find out how many objects are there for lnBrkIntvls ? can I map them to the c++ contructs or stuccts ?  Please suggest .  

thanks in advance .
Sanjay

bblanchon, 2017-03-21:
Hello Sanjay,

Have a look at:

* [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/)
* [JsonArray::size()](https://bblanchon.github.io/ArduinoJson/api/jsonarray/size/)
* [JsonArray::begin() / end()](https://bblanchon.github.io/ArduinoJson/api/jsonarray/begin_end/)

Regards,
Benoit

SanjayGautam, 2017-03-23:
Thanks blanchon , looks like the apis is almost like java , thanks I will try now .. Great work on the api

bblanchon, 2017-03-23:
You're welcome Sanjay. 
Thank you for using ArduinoJson.

closed 2017-03-23 by bblanchon