bblanchon/ArduinoJson #470: Reusing StaticJsonBuffer

andrey-gvrd, 2017-03-22:
The guide calls for an allocation of a buffer right before parsing. The problem with that is that it's easy to run out of stack space at runtime.

A preferable method would be to use a single (per file) global `StaticJsonBuffer` so that a potential memory issue is found at compile time. That approach doesn't currently work (even when clearing the buffer between parsing calls), not sure why.

bblanchon, 2017-03-22:
Hello Andrey,

Please read:

* [FAQ: Why shouldn't I use a global JsonBuffer?](https://bblanchon.github.io/ArduinoJson/faq/why-shouldnt-i-use-a-global-jsonbuffer/)
* [FAQ: How to reuse a JsonBuffer?](https://bblanchon.github.io/ArduinoJson/faq/how-to-reuse-a-jsonbuffer/)

Regards,
Benoit

andrey-gvrd, 2017-03-22:
Right on.

closed 2017-03-22 by andrey-gvrd