bblanchon/ArduinoJson #471: parsing failure of json string

holgerkoch, 2017-03-22:
Hello from Germany,

Thanks you very much for your great library ArduinoJson.

I've read the documentation and try the assistant, but something goes wrong. Maybe it is a easy thing for you to help me.

I try your library on a wemos D1.
Here is a example for the json String i got from http://api.openweathermap.org:

{"city":{"id":2929670,"name":"Erfurt","coord":{"lon":11.0333,"lat":50.9833},"country":"DE","population":203254},"cod":"200","message":1.2680583,"cnt":3,"list":[{"dt":1490180400,"temp":{"day":8,"min":4.83,"max":8,"night":4.83,"eve":5.89,"morn":8},"pressure":992.03,"humidity":84,"weather":[{"id":803,"main":"Clouds","description":"überwiegend bewölkt","icon":"04d"}],"speed":5.27,"deg":101,"clouds":68},{"dt":1490266800,"temp":{"day":7.89,"min":2.48,"max":8.95,"night":5.15,"eve":7.33,"morn":2.48},"pressure":992.33,"humidity":90,"weather":[{"id":801,"main":"Clouds","description":"ein paar Wolken","icon":"02d"}],"speed":2.91,"deg":70,"clouds":20},{"dt":1490353200,"temp":{"day":7.45,"min":4.89,"max":8.26,"night":5.7,"eve":7.35,"morn":4.89},"pressure":1001.9,"humidity":87,"weather":[{"id":500,"main":"Rain","description":"leichter Regen","icon":"10d"}],"speed":5.86,"deg":71,"clouds":88,"rain":0.34}]}

The assistant say for the buffer size and ESP8266	2009. I try it with 2200 and 3000. But now luck.

JsonObject& city = root["city"];
long city_id = city["id"]; // 2929670
const char* city_name = city["name"]; // "Erfurt"

float city_coord_lon = city["coord"]["lon"]; // 11.0333
float city_coord_lat = city["coord"]["lat"]; // 50.9833

const char* city_country = city["country"]; // "DE"
long city_population = city["population"]; // 203254

Here is my code:
bool readReponseContent() {
//  const size_t bufferSize = 3*JSON_ARRAY_SIZE(1) + JSON_ARRAY_SIZE(3) + JSON_OBJECT_SIZE(2) + 3*JSON_OBJECT_SIZE(4) + 2*JSON_OBJECT_SIZE(5) + 3*JSON_OBJECT_SIZE(6) + JSON_OBJECT_SIZE(8) + 2*JSON_OBJECT_SIZE(9) + 810;
  const size_t bufferSize = 3000;
  DynamicJsonBuffer  jsonBuffer(bufferSize);
  JsonObject& root = jsonBuffer.parseObject(client);

  if (!root.success()) {
    Serial.println("JSON parsing failed!");
    return false;
  } else {
    Serial.println("JSON parsing ok!");
  }
  strcpy(stadt.cityname, root["city"]["name"]); // "Erfurt"
  strcpy(stadt.country, root["city"]["country"]); // "DE"
  stadt.population = root["city"]["population"]; // 203254

  JsonArray& list = root["list"];
  JsonObject& list0 = list[0];
  strcpy(WeatherData[0].pressure, list0["pressure"]);
  strcpy(WeatherData[0].humidity, list0["humidity"]);
  JsonObject& list0_temp = list0["temp"];
  strcpy(WeatherData[0].tempDay, list0_temp["day"]);
  strcpy(WeatherData[0].tempMin, list0_temp["min"]);
  strcpy(WeatherData[0].tempMax, list0_temp["max"]);
  JsonObject& list0_weather0 = list0["weather"][0];
  strcpy(WeatherData[0].description, list0_weather0["description"]);
  strcpy(WeatherData[0].icon, list0_weather0["icon"]);
........

        if (readReponseContent()) {
          
          Serial.println(stadt.cityname);
          Serial.println(stadt.country);
          Serial.println(stadt.population);
          for (int i=0; i<1; i++) {
            Serial.println(i);
            Serial.println(WeatherData[i].tempDay);
            Serial.println(WeatherData[i].description);
            Serial.println(WeatherData[i].pressure);
            Serial.println(WeatherData[i].icon);            
          }

And here is the output from the console:
JSON parsing ok!
7.89                     should be Erfurt
7.89                     should be DE 
775422008          should be 203254
0                          correct
9.27                     correct
überwiegend bewölkt  correct
992.03                         correct
04d7.89                       should 

It is possible, that the json string is to complex/long?

Thank you very much for every hint.

Best regards

Holger

holgerkoch, 2017-03-22:
sometimes i got this error in the console:

7.89
7.89
775422008
0
9
überwiegend bew��lkt
992.03
04d7.89
Disconnect

Exception (28):
epc1=0x4020427e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00618000 depc=0x00000000

ctx: cont 
sp: 3ffefdb0 end: 3ffeffe0 offset: 01a0

>>>stack>>>
3ffeff50:  3fff1000 00000009 3ffe8580 3ffeefb4  
3ffeff60:  3ffe84a3 0000000a 3ffeefc0 3ffee948  
3ffeff70:  3fffdad0 3ffeefb4 72616170 4020429c  
3ffeff80:  3ffe8577 3ffee948 00000000 4020211a  
3ffeff90:  3fffdad0 3ffee948 3ffee954 40202707  
3ffeffa0:  3fffdad0 3ffee948 3ffeef00 40202741  
3ffeffb0:  3fffdad0 3ffee948 3ffeef00 40203d5d  
3ffeffc0:  feefeffe 00000000 3ffeefac 40204d10  
3ffeffd0:  feefeffe feefeffe 3ffeefc0 40100718  
<<<stack<<<
H��R8���

bblanchon, 2017-03-22:
Hello Holger,

I don't think the input is too big or too complex.
The ESP8266 has a sufficient amount of memory and [you're not the first to try](https://www.youtube.com/watch?v=C2LBqUB7g3g).
Moreover, the `success()` function returns `true` meaning that the parse succeeds.

I believe the mistake is in the way you store the data.
Can you show the definition of `stadt`?
Better, can you create a [Gist](https://gist.github.com/) with the complete program?

Regards,
Benoit

holgerkoch, 2017-03-22:
Hello Benoit,

thank you for your fast responds.
Maybe you are right, because i found a lot of working examples, also with http://api.openweathermap.org. But where is my fault?

Here is the definition of both data structur:

 struct varcity {
  char cityname[];
  char country[5];
  long population;
} stadt;

//weather Data
struct weather {
  char tempDay[5];
  char tempMin[5];  
  char tempMax[5];
  char pressure[7];
  char humidity[2];
  char description[25];
  char icon[3];
};

weather WeatherData[2];

The complete code is under:  

https://gist.github.com/holgerkoch/341b783a0a80e38764b4ea9f532a6c86

best regards

Holger

bblanchon, 2017-03-22:
Hello Holger,

You forgot to specify the size of `cityname` in the `varcity` declaration.
You need to replace with something like:

```c++
struct varcity {
    char cityname[32];
    char country[32];
    long population;
} stadt;
```

Regards,
Benoit

holgerkoch, 2017-03-22:
Hello Benoit,

thank you for your hint. I changed the definition to

struct varcity {
  char cityname[32];
  char country[32];
  long population;
} stadt;

since them, the wemos don't crash anymore, but the output is further strange:

JSON parsing ok!
6.85
 Regen
203254
0
6
leichter Regen
991.33
10d6.85

Do you have another hint for me?

Best regards

Holger

bblanchon, 2017-03-23:
Hello Holger,

Fields in the `weather` structure are too small to hold the strings.
For instance, `icon` is only 3 chars, but 4 are required to store "10d" with the null-terminator.

If you're uncomfortable using C strings, you can use the `String` class, like this:

```c++
struct varcity {
  String cityname;
  String country;
  long population;
} stadt;
```

then you don't need to `strcpy()`, it's done by the `String` class:

```c++
stadt.cityname = root["city"]["name"].as<char*>();
stadt.country, root["city"]["country"].as<char*>();
stadt.population = root["city"]["population"];
```

Regards,
Benoit

holgerkoch, 2017-03-23:
Hello Benoit,

thank you for this hint. I forgot that i need i further character for the null-terminator. So my first try was to increase every variable. That was the right way. Now are all weather information (WeatherData) correct.
Unfortunately there is still a problem with the "varcity". If i try your suggestion, i got this:

struct varcity {
  String cityname;
  String country;
  long population;
} stadt;
:
:
JsonObject& city = root["city"];
stadt.cityname = root["city"]["name"].as<char*>();
stadt.country, root["city"]["country"].as<char*>();
stadt.population = root["city"]["population"];

JSON parsing ok!
Soft WDT reset

ctx: cont 
sp: 3ffefdd0 end: 3fff0000 offset: 01b0

>>>stack>>>
3ffeff80:  3ffeff94 3ffee970 3ffe0000 3ffeff94  
3ffeff90:  3ffeff0a 00000915 3632374c 40205140  
3ffeffa0:  00001f40 00000000 3ffee970 3ffeefcc  
3ffeffb0:  3fffdad0 3ffeef18 3ffeef18 402051bd  
3ffeffc0:  0a0d0a0d 00000100 000001f4 4020436d  
3ffeffd0:  3fffdad0 3ffeef18 3ffee954 40203da3  
3ffeffe0:  3fffdad0 00000000 3ffeefc4 40204dd4  
3ffefff0:  feefeffe feefeffe 3ffeefe0 40100718  
<<<stack<<<

My next try was 
  JsonObject& city = root["city"];
  strcpy(stadt.cityname, city["name"]); // "Erfurt"
  strcpy(stadt.country, city["country"]); // "DE"
  stadt.population = city["population"]; // 203254

JSON parsing ok!
7.26
erwiegend bewölkt
203254

The population Value is correct. But cityname and country are strange twisted. But the parsing is correct. I insert the following line:

Serial.println(root["city"]["name"].as<char*>());

and got

JSON parsing ok!
Erfurt

Perfekt. Unfortunately i can't assign this to a variable.

  Serial.println(root["city"]["name"].as<char*>());
  stadt.cityname = root["city"]["name"].as<char*>();

and i got 

Soft WDT reset

ctx: cont 
sp: 3ffefdb0 end: 3ffeffe0 offset: 01b0

>>>stack>>>
3ffeff60:  3ffeff74 3ffee954 3ffe0000 3ffeff74  
3ffeff70:  3ffeff0a 0000099e 363237d5 402050a4  
3ffeff80:  00001f40 00000000 3ffee954 3ffeefb0  
3ffeff90:  3fffdad0 3ffeeefc 3ffeeefc 40205121  
3ffeffa0:  0a0d0a0d 00000100 000001f4 402042d1  
3ffeffb0:  3fffdad0 3ffeeefc 3ffee924 40203d07  
3ffeffc0:  3fffdad0 00000000 3ffeefa8 40204d38  
3ffeffd0:  feefeffe feefeffe 3ffeefc0 40100718  
<<<stack<<<

Do you have another hint for me?

Best regards

Holger

bblanchon, 2017-03-24:
Hello Holger,

I found another mistake in the gist.
You declare `WeatherData` like this:

```c++
weather WeatherData[2];
```

But in the code, you use 3 slots of the array 💣 
You need to increase the size.

Regards,
Benoit

holgerkoch, 2017-03-24:
Hello Benoit,

thank you very much. You make my day, that was the point. I thought, array start with 0, so WeatherData[2]; means 0,1 and 2. To my excuse, my last programming experience with c are nearly 20 years ago.

Have a nice weekend

Holger

bblanchon, 2017-03-24:
You're welcome Holger. 
Thank you for using ArduinoJson.

closed 2017-03-24 by bblanchon