bblanchon/ArduinoJson #472: Problems parsing response body

HansWurst90, 2017-03-24:
Hi,
I'm trying to parse the body of a HTTP-Response but can't get it to work.
my code is:

```c++
bool getStatus(String accessory){
    // "{ \"characteristics\": [{ \"aid\": 10, \"iid\": 9, \"value\": true, \"status\": 0}] }";
    // {"characteristics":[{"aid":10,"iid":9,"value":true,"status":0}]}
    Serial.println();
    Serial.println("Application>\tStart of GET");
    http_request_t getRequest = request;
    getRequest.path = "/characteristics?id="+ accessory +".9";
    http.get(getRequest, response, headers);
    char charBuf[300];
    response.body.toCharArray(charBuf, 300);
    char json[] = {};
    strcpy(json, charBuf);
    StaticJsonBuffer<300> jsonBuffer;
    JsonObject& root = jsonBuffer.parseObject(json);
    if (!root.success())
    {
      Serial.println("parseObject() failed");
    }
    const bool state = root["characteristics"]["value"];
    Serial.print("Application>\tResponse status: ");
    Serial.println(response.status);
    Serial.print("Application>\tResponse state: ");
    Serial.println(state);
    Serial.print("Application>\tHTTP Response Body: ");
    Serial.println(response.body);
    // return state;
    return false;
}
```

and my Serial output is:

```
Application>    Start of GET
Application>    Response status: 207
Application>    Response state: 0
Application>    HTTP Response Body: {"characteristics":[{"aid":10,"iid":9,"value":true,"status":0}]}
```

so `root["characteristics"]["value"];` should deliver `true`, but it prints `0` instead.
I'm am not very familiar with c++ yet and kind if clueless right now.

bblanchon, 2017-03-24:
Hi Jan,

Have a look at:

* [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/)
* [FAQ: Parsing succeeds but I can't read the values!](https://bblanchon.github.io/ArduinoJson/faq/parsing-succeeds-but-i-cant-read-the-values/)
* [FAQ: Why parsing fails?](https://bblanchon.github.io/ArduinoJson/faq/why-parsing-fails/)

Regards,
Benoit

bblanchon, 2017-04-01:
Did you try `root["characteristics"][0]["value"];` ?

HansWurst90, 2017-04-08:
Sry, I didn't have time to try yet.
I'll let you know about my result.

bblanchon, 2017-04-18:
Hi @HansWurst90.
I'm closing this issue, please reopen later if you need.

closed 2017-04-18 by bblanchon