bblanchon/ArduinoJson #473: Parsing fails after calling the function a number of times

ghislainborremans, 2017-03-25:
Hello,
Below you find the code ( udp_read) 
this function is called and reads udp input. 
after sending 7 times the "LogIp"string (see belowà , the parser returns failed and stays failed. even if the string is correct json. 
any ideas what i am doing wrong? 

best regards 

Ghislain 

/*
 WiFi UDP Send and Receive String

 */

#include <SPI.h>
#include <WiFiManager.h>

int my_inet_aton(const char* aIPAddrString, IPAddress& aResult);

//these are the strings that are sent via UDP
//after calling 7 times, the parsing stops working and returns "failed".)
//
//{"LogIP":"192.168.1.33","LogPort":8082}
//{"Taak":"Log"}
//below is an invalid string used to check this as well
//TEST

void udp_read() {
	//Serial.println("in udp_read at");
	//Serial.println(millis());
	// if there's data available, read a packet
	int packetSize = Udp.parsePacket();
	if (packetSize) {
		Serial.print("Received packet of size ");
		Serial.println(packetSize);
		Serial.print("From ");
		remoteIp = Udp.remoteIP();
		Serial.print(remoteIp);
		remotePort = Udp.remotePort();
		Serial.print(", port ");
		Serial.println(remotePort);

		// read the packet into packetBufffer
		packetBuffer[0] = 0;
		int len = Udp.read(packetBuffer, 255);

		if (len > 0) {
			packetBuffer[len] = 0;
		}
		Serial.println(packetBuffer);

		// Root of the object tree.
		//
		// It's a reference to the JsonObject, the actual bytes are inside the
		// JsonBuffer with all the other nodes of the object tree.
		// Memory is freed when jsonBuffer goes out of scope.
		JsonObject& root = jsonBuffer.parseObject(packetBuffer);

		// Test if parsing succeeds.
		if (!root.success()) {
			Serial.println("parseObject() failed");
			//	return;
		} else {
			jsonTaak = root["Taak"];
			Serial.println(root.containsKey("LogIP"));
			if (root.containsKey("LogIP")) {
				Serial.print("LogPort");
				jsonLogPort = root["LogPort"];
				Serial.println(jsonLogPort);
				Serial.print("LogIpStr ");
				jsonLogIpStr = root["LogIP"];
				Serial.println(jsonLogIpStr);
				my_inet_aton(jsonLogIpStr, jsonLogIP);
				Serial.print("LogIp ");
				Serial.println(jsonLogIP);
			};
			Serial.println("na IP:");
			Serial.println(jsonTaak);
		}

		sendlog();

		// send a reply, to the IP address and port that sent us the packet we received
		Udp.beginPacket(remoteIp, remotePort);
		Udp.write(ReplyBuffer);
		Udp.write(packetBuffer);
		Udp.endPacket();

		Serial.println("voor log");
		if (jsonTaak != 0) {
			if (!strcmp(jsonTaak, "Log")) {
				Serial.println("voor log");
				jsonLog = 'T';
				sendout("NoTime", "readUdp", "Log True ", 0, "CR", "UDP");
			}
			if (!strcmp(jsonTaak, "NoLog")) {
				jsonLog = 'F';
				sendout("NoTime", "readUdp", "Log False ", 0, "CR", "UDP");
			}
		}

	}

}

bblanchon, 2017-03-25:
Hi Ghislain,

Have a look at:

* [FAQ: The first parsing succeeds, why do the next ones fail?](https://bblanchon.github.io/ArduinoJson/faq/the-first-parsing-succeeds-why-do-the-next-ones-fail/)
* [FAQ: Why shouldn't I use a global JsonBuffer?](https://bblanchon.github.io/ArduinoJson/faq/why-shouldnt-i-use-a-global-jsonbuffer/)
* [FAQ: How to reuse a JsonBuffer?](https://bblanchon.github.io/ArduinoJson/faq/how-to-reuse-a-jsonbuffer/)

Regards,
Benoit

ghislainborremans, 2017-03-27:
Hello
it is indeed the fact that it is a global variable. I changed it and now it is ok. 
thanks

closed 2017-03-27 by ghislainborremans