bblanchon/ArduinoJson #474: Parsing succeeds but I am getting 0 on my serial output

mpcpro, 2017-03-25:
Hi. 

I got my script working and my ESP8266 is able to get the information from openweathermap fine. The string of texts shows up fine. However when calling specific information in the serial console I am getting for example these readings:

Current Temperature: 0
Rain: 0
Wind: 0

Example:

Even though the data parsed has information:
Json: ......   "main":{"temp":4.07,"temp_min":4.07,"temp_max":4.07,   ........

Code: 

Serial.print("Current Temperature: ");
Serial.print((double)root["main"]["temp"]);
Serial.println(" K\n");

Output on serial monitor:

Current Temperature: 0 K

I'm wondering what I am doing wrong. Since I am unable to see the problem.

File:

[KindaworksOpenWeaterMap.txt](https://github.com/bblanchon/ArduinoJson/files/870373/KindaworksOpenWeaterMap.txt)

Thanks.

- MPC

bblanchon, 2017-03-27:
Hi MPC,

Did you mean `root["list"][0]["main"]["temp"]`?

See:

* [FAQ: Parsing succeeds but I can't read the values!](https://bblanchon.github.io/ArduinoJson/faq/parsing-succeeds-but-i-cant-read-the-values/)
* [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/)

*(Sorry for the long delay, I was sure I answer to your question two days ago but I probably forgot to click on the button.)*

Regards,
Benoit

mpcpro, 2017-03-27:
Damn, I forgot to check out that FAQ. Thanks. I'll give it a go when I get home from work.

mpcpro, 2017-03-27:
Thanks! That worked perfectly!

bblanchon, 2017-03-27:
You're welcome MPC. 
Thank you for using ArduinoJson.

closed 2017-03-27 by bblanchon