bblanchon/ArduinoJson #476: Escaped strings in JSON

vossilius, 2017-03-29:
Hi,

I have problems similar to https://github.com/bblanchon/ArduinoJson/issues/16 when building a JSON. 
To exemplify, take the JsonGeneratorExample and add `data.add("This is a \"String\"");` below line 48 resulting in:

```
{
  "sensor": "gps",
  "time": 1351824120,
  "data": [
    48.756080,
    2.302038,
    "This is a \"String\""
  ]
}
```

In my code I use a String containing something like 
```
String payload= "{";
  payload += "\"Temperature [*C]\" : ";
  payload += sensor.temp;
 payload += "}";
```
which I try to pass into a JSON object in another function:
```
...
  root["d"] = payload;
```
Yet, the results is:
```
{
...
"d" : "{\"Temperature [*C]\" :  23.4}
"}
```
ArduinoJSON v5.8.4 is installed.

How do I escape the double-quote when passing a string into the JSON object?

bblanchon, 2017-03-29:
Hello @vossilius,

Why don't you create a nested object in the array?

```c++
JsonObject& payload = root.createNestedObject("d");
payload["Temperature [*C]"] = sensor.temp;
```

You can prevent string escaping by wrapping a `const char*` in a `RawJson`, but I don't think that's the right solution for you.

Regards,
Benoit

vossilius, 2017-03-31:
Hello Benoit,

I was thinking of option 1 but this would rather break the idea of the function I wrote to build a static JSON message header and add in the payload whenever calling the function.

Could you elaborate option 2 a bit more?

But shouldn't the string escaping work in your function anyway for the ease of use?

Thanks and Regards,
vossilius

vossilius, 2017-03-31:
Ok. I went for option 1 passing the entire object as argument to the function.
Works great but there is one odd effect, though. When I do a prettyPrint the nested JSON object shows top even though I appended it last.

bblanchon, 2017-03-31:
About option 2, see [FAQ: How to prevent string escaping?](https://bblanchon.github.io/ArduinoJson/faq/how-to-prevent-string-escaping/)

closed 2017-04-08 by bblanchon